platformio/platformio-vscode-ide #3668: Start PIO Home Server

gseb45, 2023-04-01:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.77.0
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\platformio\cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\click\core.py", line 1654, in invoke
    super().invoke(ctx)
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\platformio\__main__.py", line 56, in cli
    maintenance.on_platformio_start(ctx, caller)
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 42, in on_platformio_start
    after_upgrade(ctx)
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 154, in after_upgrade
    cleanup_content_cache("http")
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\platformio\cache.py", line 164, in cleanup_content_cache
    with ContentCache(namespace) as cc:
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\platformio\cache.py", line 36, in __enter__
    self.delete()
  File "C:\Users\Sebastien GITTON\.platformio\penv\lib\site-packages\platformio\cache.py", line 132, in delete
    with open(self._db_path, mode="w", encoding="utf8") as fp:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\Sebastien GITTON\\.platformio\\.cache\\http\\db.data'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\Sebastien GITTON\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10645
	at ChildProcess.o (c:\Users\Sebastien GITTON\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

ivankravets, 2023-05-10:
Please remove `C:\\Users\\Sebastien GITTON\\.platformio\\.cache` folder and restart VSCode.

gseb45, 2023-05-10:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN

    https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue

%23 Configuration

VSCode: 1.78.1
PIO IDE: v3.1.1
System: Windows_NT, 10.0.19044, x64

%23 Exception

Error: Error: The directory `C:\Users\Sebastien GITTON\.platformio` or its parent directory is not owned by the current user and PlatformIO can not store configuration data.
Please check the permissions and owner of that directory.
Otherwise, please remove manually `C:\Users\Sebastien GITTON\.platformio` directory and PlatformIO will create new from the current user.

  at c:\Users\Sebastien GITTON\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10645
  at ChildProcess.o (c:\Users\Sebastien GITTON\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
  at ChildProcess.emit (node:events:513:28)
  at ChildProcess.emit (node:domain:489:12)
  at maybeClose (node:internal/child_process:1112:16)
  at ChildProcess._handle.onexit (node:internal/child_process:304:5)

ivankravets, 2023-05-11:
Did you use Admin rights to install PlatformIO Core? No need to use Admin rights. Remove `C:\Users\Sebastien GITTON\.platformio` folder and restart VSCode.