platformio/platformio-vscode-ide #3673: Cannot open Terminal

max5555, 2023-04-07:
I cannot open a new terminal in platformio. I made clean installation of VSCode (v 1.77.0), PlatformIO Core, version 6.1.5a3, Windows 10x64. What is the the reason for the following error?

```
-noexit : The term '-noexit' is not recognized as the name of a cmdlet, function, script file, or operable program.
Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ -noexit -command try { . c:\Users\y\AppData\Local\Programs\Microsoft  ...
+ ~~~~~~~
+ CategoryInfo      	: ObjectNotFound: (-noexit:String) [], CommandNotFoundException
+ FullyQualifiedErrorId : CommandNotFoundException

```

After this message the terminal is collapsing.

ivankravets, 2023-04-07:
It seems you misconfigured terminal settings in the VSCode. Please open VSCode settings and remove any custom settings for the VSCode's Terminal.

closed 2023-04-07 by ivankravets

michelebombardi, 2023-09-09:
@max5555 i'm experiencing the same issue and I have no custom configurations. It give me this error after a fresh installation of VS Code. How did you solve this?