platformio/platformio-vscode-ide #3681: VS Code PIO menu items do not work

justoke, 2023-04-15:
I have been working with PlatformIO for a number of years, just this last week one of my long time projects exhibits a frustrating condition. None of the PIO menus and task menu icons will actually fun any of the commands needed to build, clean and upload my project. Only using the command line in the Terminal works. All menus are available, but whether using the command palette or any of the PIO menus they just don't do anything.

[image]

I have tried, renaming .pio folder and re-opening project. Closing VS Code, restarting my PC. I just can't figure out why this may be happening. I have other projects that work just fine. I found some suggestions regarding resetting the workspace but I can't find the menu or option to do so.

Any suggestions greatly appreciated.

ivankravets, 2023-04-18:
Do you use the latest VSCode + PlatformIO IDE for the VSCode extension? Could you try to disable all extensions and keep ONLY `C/C++` + `PlatformIO IDE`. Restart VSCode.

Does it work now?

justoke, 2023-04-18:
That was it! I disabled a whole set of CMAKE extensions and now it works as before.

closed 2023-04-18 by justoke