platformio/platformio-vscode-ide #3686: Already up-to-date. Updating metadata for the vscode IDE...

nglessner, 2023-04-19:
Message:

PlatformIO: Configuring project: Already up-to-date.
Updating metadata for the vscode IDE...

Problem: Platform IO is unable to be used until this "completes" or times out. This takes multiple minutes.

Expectation: to be able to use the IDE within a few seconds of starting visual studio code.

ivankravets, 2023-04-19:
Could you share your `platformio.ini` file?

nglessner, 2023-04-19:
It does, it just takes forever "Scanning Dependencies".  I've played around with lib_ldf_mode and turing it off fixes both the Scanning Dependencies and Updating Metadata slowness - but then of course I have to have _every_ library defined.

ivankravets, 2023-04-19:
Could you provide a simple project to reproduce this issue?

nglessner, 2023-04-19:
It's basically this project: https://github.com/lmarzen/esp32-weather-epd/tree/main/platformio

ivankravets, 2023-04-22:
It seems you have issues with your Git configuration. Please use official configuration for that project https://github.com/lmarzen/esp32-weather-epd/blob/main/platformio/platformio.ini

It does not use an Git URLs. Git can ask SSH keys, passwords and etc. This is the reason why it freezes in subprocess.

closed 2023-04-22 by ivankravets

cgudrian, 2023-05-12:
I'm having the same issues with the above project. I rather believe it's the large number of header files (almost 10,000) which imposes quite some work on the dependency resolver.

nglessner, 2023-05-12:
@cgudrian have you tried cutting out library files that are not used?  I haven't had time recently to look at this project again, but that was my next attempt.

cgudrian, 2023-05-12:
It's include files like [this one](https://github.com/lmarzen/esp32-weather-epd/blob/main/platformio/include/icons/icons_128x128.h) which are problematic, not libraries.

The obvious workaround would be to only pick the specific subset of icons and fonts that are actually used (instead of having the linker dispose of them).

Since I've compiled and uploaded that project only a couple of times I personally can live with the current behavior and have not spent any time trying to optimize build times.