platformio/platformio-vscode-ide #3694: IDE sync popup during build makes the build fail

NathanSweet, 2023-04-27:
Many times I run a build and it fails. I run it again and it succeeds. I noticed that the VS Code project sync dialog flashes briefly when this occurs. I think a sync during the build breaks the build. Build failure looks like:

```
framework-arduinoespressif32/cores/esp32/main.cpp:72:1: fatal error: opening dependency file .pio/build/master/FrameworkArduino/main.cpp.d: No such file or directory
```

For example, if I edit platformio.ini and press ctrl+B,enter to build, a second or so into the build the project syncs and the build fails. ctrl+B,enter again and the build succeeds.

I think a sync should be done before a build and syncing should be disallowed during a build.

In case it matters, I’m using:
```
lib_ldf_mode = off
lib_compat_mode = off
```
I’m editing my platformio.ini to add a dependency to lib_deps.

Possibly related: #3207

ivankravets, 2023-05-10:
Please forward to https://community.platformio.org/

NathanSweet, 2023-05-10:
How do I forward an issue to a forum? Are you implying this is not actually an issue but is instead a question? I would disagree.

I originally posted the issue on the forum and was directed to the issue tracker. It seems we're in loop!
https://community.platformio.org/t/sync-popup-during-build-makes-the-build-fail/33468

The issue can be avoided relatively easily, once the user figures out WTF is going on, but the symptoms make for quite bad UX: builds appear to fail for bizarre reasons, yet work at other times.

ivankravets, 2023-05-10:
> project syncs 

What do you mean? We don't provide any synchronization. Could you disable ALL VSCode 3rd party extensions and keep only C/C++ and PlatformIO IDE? Could you reproduce this issue again?

NathanSweet, 2023-05-10:
By "sync" I mean whatever is happening when this popup is shown, a few seconds after saving `platformio.ini`:
[image]

Yes, it can be reproduced easily and every time. Edit `platformio.ini` without saving, then run a build (`ctrl+alt+B`). The `platformio.ini` gets saved and the build starts, a few seconds later the popup appears, then the build fails.

The workaround is of course: don't do that, but maybe it can be handled another way. Ideally the build could delay until any pending "sync" is complete. Maybe when a sync occurs it can check if a build is in progress, then if that build fails print a message saying the failure may be caused by the sync. Basically any way to lessen the surprise of a build failing seemingly sporadically would be beneficial.

I don't use any 3rd party extensions, only VS Code + PIO IDE. I installed VS Code specifically for PIO, so it's a fresh install.

ivankravets, 2023-05-10:
Thanks for the explanation. I marked this issue as "known".  We will think of a solution for it. When you edit `platformio.ini` and make changes, we remove a build cache. The issue arises in parallel when updating metadata tries to reach non-existing files.