platformio/platformio-vscode-ide #3695: Cannot create Zephyr/nrf51 Project with Space in Project Name

kerryland, 2023-04-29:
To reproduce in PIO 3.1.1:

1. Create a new project named "Whitespace Bug" for a "Nordic nRF51X22" using Zephyr RTOS.
2. Click "Finish"
3. See "Could not initialize project".

No such error occurs when there is no whitespace in the project name.

Full [error log](https://github.com/platformio/platformio-vscode-ide/files/11358575/platformio-error.log) attached. I even made it readable by fixing the EOL characters :-)

[image]
[image]

mrengineer7777, 2023-05-02:
Spaces in project name or path break a lot of development tools.  Best practice is to use an underscore instead of a space.  Save yourself the headache.

ivankravets, 2023-05-10:
Thanks for the provided help.

closed 2023-05-10 by ivankravets