platformio/platformio-vscode-ide #3699: i installed platformIO but failed such as the following

Nungki1999, 2023-05-05:
Error:   error: subprocess-exited-with-error
  
  python setup.py bdist_wheel did not run successfully.
  exit code: 1
  
  [417 lines of output]
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib
  creating build\lib\platformio
  copying platformio\app.py -> build\lib\platformio
  copying platformio\cache.py -> build\lib\platformio
  copying platformio\cli.py -> build\lib\platformio
  copying platformio\compat.py -> build\lib\platformio
  copying platformio\exception.py -> build\lib\platformio
  copying platformio\fs.py -> build\lib\platformio
  copying platformio\http.py -> build\lib\platformio
  copying platformio\maintenance....
  
  
  this my error ini errror saya saat install PlatformIO IDE

ivankravets, 2023-05-10:
See https://github.com/platformio/platformio-core-installer/issues/2894

closed 2023-05-10 by ivankravets

Nungki1999, 2023-05-29:
I have tried the steps shown but it still doesn't work

Nungki1999, 2023-05-29:
Error:   error: subprocess-exited-with-error
  
  python setup.py bdist_wheel did not run successfully.
  exit code: 1
  
  [425 lines of output]
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib
  creating build\lib\platformio
  copying platformio\app.py -> build\lib\platformio
  copying platformio\cache.py -> build\lib\platformio
  copying platformio\cli.py -> build\lib\platformio
  copying platformio\compat.py -> build\lib\platformio
  copying platformio\exception.py -> build\lib\platformio
  copying platformio\fs.py -> build\lib\platformio
  copying platformio\http.py -> build\lib\platformio
  copying platformio\maintenance....