platformio/platformio-vscode-ide #3707: Start PIO Home Server

botam4, 2023-05-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.78.1
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.25336, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "runpy.py", line 197, in _run_module_as_main
  File "runpy.py", line 87, in _run_code
  File "C:\Users\ionco\.platformio\penv\lib\site-packages\platformio\__main__.py", line 19, in <module>
    import click
  File "C:\Users\ionco\.platformio\penv\lib\site-packages\click\__init__.py", line 7, in <module>
    from .core import Argument as Argument
  File "C:\Users\ionco\.platformio\penv\lib\site-packages\click\core.py", line 16, in <module>
    from . import types
  File "C:\Users\ionco\.platformio\penv\lib\site-packages\click\types.py", line 8, in <module>
    from ._compat import _get_argv_encoding
  File "C:\Users\ionco\.platformio\penv\lib\site-packages\click\_compat.py", line 518, in <module>
    from ._winconsole import _get_windows_console_stream
  File "C:\Users\ionco\.platformio\penv\lib\site-packages\click\_winconsole.py", line 13, in <module>
    from ctypes import byref
  File "<frozen zipimport>", line 259, in load_module
  File "ctypes\__init__.py", line 8, in <module>
ImportError: Module use of python310.dll conflicts with this version of Python.

	at c:\Users\ionco\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10645
	at ChildProcess.o (c:\Users\ionco\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1112:16)
	at ChildProcess._handle.onexit (node:internal/child_process:304:5)
  ```

ivankravets, 2023-05-31:
Please remove `C:\Users\ionco\.platformio\penv` folder and restart VSCode.

botam4, 2023-05-31:
> Please remove `C:\Users\ionco\.platformio\penv` folder and restart VSCode.

Thanks, Btw What have u been doing lately? i have opened this issue 3 weeks ago