platformio/platformio-vscode-ide #3712: PIO for VSCode Installation Error

cncherisher, 2023-05-16:
Description of problem
follow this tutorial
https://docs.platformio.org/en/latest/integration/ide/vscode.html#id1

System configuration
Version: 1.78.2 (system setup)
Date: 2023-05-10T14:39:26.248Z
Electron: 22.5.2
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Windows_NT x64 10.0.19045
PIO IDE: v3.1.1

Exception
```
Error:   error: subprocess-exited-with-error      python setup.py bdist_wheel did not run successfully.   exit code: 1      [435 lines of output]   running bdist_wheel   running build   running build_py   creating build   creating build\lib   creating build\lib\platformio   copying platformio\app.py -> build\lib\platformio   copying platformio\cache.py -> build\lib\platformio   copying platformio\cli.py -> build\lib\platformio   copying platformio\compat.py -> build\lib\platformio   copying platformio\exception.py -> build\lib\platformio   copying platformio\fs.py -> build\lib\platformio   copying platformio\http.py -> build\lib\platformio   copying platformio\maintenance....
```
```
Error:   error: subprocess-exited-with-error
  
  python setup.py bdist_wheel did not run successfully.
  exit code: 1
  
  [435 lines of output]
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib
  creating build\lib\platformio
  copying platformio\app.py -> build\lib\platformio
  copying platformio\cache.py -> build\lib\platformio
  copying platformio\cli.py -> build\lib\platformio
  copying platformio\compat.py -> build\lib\platformio
  copying platformio\exception.py -> build\lib\platformio
  copying platformio\fs.py -> build\lib\platformio
  copying platformio\http.py -> build\lib\platformio
  copying platformio\maintenance.py -> build\lib\platformio
  copying platformio\proc.py -> build\lib\platformio
  copying platformio\public.py -> build\lib\platformio
  copying platformio\telemetry.py -> build\lib\platformio
  copying platformio\util.py -> build\lib\platformio
  copying platformio\__init__.py -> build\lib\platformio
  copying platformio\__main__.py -> build\lib\platformio
  creating build\lib\platformio\account
  copying platformio\account\cli.py -> build\lib\platformio\account
  copying platformio\account\client.py -> build\lib\platformio\account
  copying platformio\account\validate.py -> build\lib\platformio\account
  copying platformio\account\__init__.py -> build\lib\platformio\account
  creating build\lib\platformio\builder
  copying platformio\builder\main.py -> build\lib\platformio\builder
  copying platformio\builder\__init__.py -> build\lib\platformio\builder
  creating build\lib\platformio\check
  copying platformio\check\cli.py -> build\lib\platformio\check
  copying platformio\check\defect.py -> build\lib\platformio\check
  copying platformio\check\__init__.py -> build\lib\platformio\check
  creating build\lib\platformio\commands
  copying platformio\commands\boards.py -> build\lib\platformio\commands
  copying platformio\commands\ci.py -> build\lib\platformio\commands
  copying platformio\commands\lib.py -> build\lib\platformio\commands
  copying platformio\commands\platform.py -> build\lib\platformio\commands
  copying platformio\commands\settings.py -> build\lib\platformio\commands
  copying platformio\commands\update.py -> build\lib\platformio\commands
  copying platformio\commands\upgrade.py -> build\lib\platformio\commands
  copying platformio\commands\__init__.py -> build\lib\platformio\commands
  creating build\lib\platformio\debug
  copying platformio\debug\cli.py -> build\lib\platformio\debug
  copying platformio\debug\exception.py -> build\lib\platformio\debug
  copying platformio\debug\helpers.py -> build\lib\platformio\debug
[... 444 lines cut ...]
Collecting idna<4,>=2.5 (from requests==2.*->platformio)
  Using cached idna-3.4-py3-none-any.whl (61 kB)
Collecting certifi>=2017.4.17 (from requests==2.*->platformio)
  Using cached certifi-2023.5.7-py3-none-any.whl (156 kB)
Collecting anyio<5,>=3.4.0 (from starlette==0.26.*->platformio)
  Using cached anyio-3.6.2-py3-none-any.whl (80 kB)
Collecting typing-extensions>=3.10.0 (from starlette==0.26.*->platformio)
  Using cached typing_extensions-4.5.0-py3-none-any.whl (27 kB)
Collecting h11>=0.8 (from uvicorn==0.22.*->platformio)
  Using cached h11-0.14.0-py3-none-any.whl (58 kB)
Collecting sniffio>=1.1 (from anyio<5,>=3.4.0->starlette==0.26.*->platformio)
  Using cached sniffio-1.3.0-py3-none-any.whl (10 kB)
Building wheels for collected packages: platformio
  Building wheel for platformio (setup.py): started
  Building wheel for platformio (setup.py): finished with status 'error'
  Running setup.py clean for platformio
Failed to build platformio

	at c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5303
	at ChildProcess.o (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1112:16)
	at ChildProcess._handle.onexit (node:internal/child_process:304:5)
```

cncherisher, 2023-05-16:
emmm, seems to duplicate with #3699
Enabling Windows Long Path solve this problem
https://www.microfocus.com/documentation/filr/filr-4/filr-desktop/t47bx2ogpfz7.html

closed 2023-05-16 by cncherisher

Nungki1999, 2023-05-29:
I have tried the steps shown but it still doesn't work

lo1414ful, 2024-06-15:
> emmm, seems to duplicate with #3699 Enabling Windows Long Path solve this problem https://www.microfocus.com/documentation/filr/filr-4/filr-desktop/t47bx2ogpfz7.html

I also followed the steps mentioned here to enable long paths, but with no success (deleting .platformio after every try).
I then found an [article](https://www.howtogeek.com/266621/how-to-make-windows-10-accept-file-paths-over-260-characters/) which uses regedit to activate windows long path. After restarting and deleting .platformio once more, it finally worked.