platformio/platformio-vscode-ide #3721: Start PIO Home Server

ShadyEgypt, 2023-05-22:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.78.2
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.21996, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "C:\Users\shady\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "C:\Users\shady\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\shady\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\shady\.platformio\penv\lib\site-packages\platformio\cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "C:\Users\shady\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\shady\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\shady\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\shady\.platformio\penv\lib\site-packages\platformio\home\cli.py", line 91, in cli
    run_server(
  File "C:\Users\shady\.platformio\penv\lib\site-packages\platformio\home\run.py", line 87, in run_server
    uvicorn.run(
  File "C:\Users\shady\.platformio\penv\lib\site-packages\uvicorn\main.py", line 569, in run
    server.run()
  File "C:\Users\shady\.platformio\penv\lib\site-packages\uvicorn\server.py", line 60, in run
    return asyncio.run(self.serve(sockets=sockets))
  File "C:\Users\shady\.platformio\python3\lib\asyncio\runners.py", line 39, in run
    loop = events.new_event_loop()
  File "C:\Users\shady\.platformio\python3\lib\asyncio\events.py", line 761, in new_event_loop
    return get_event_loop_policy().new_event_loop()
  File "C:\Users\shady\.platformio\python3\lib\asyncio\events.py", line 659, in new_event_loop
    return self._loop_factory()
  File "C:\Users\shady\.platformio\python3\lib\asyncio\windows_events.py", line 315, in __init__
    super().__init__(proactor)
  File "C:\Users\shady\.platformio\python3\lib\asyncio\proactor_events.py", line 629, in __init__
    self._make_self_pipe()
  File "C:\Users\shady\.platformio\python3\lib\asyncio\proactor_events.py", line 760, in _make_self_pipe
    self._ssock, self._csock = socket.socketpair()
  File "C:\Users\shady\.platformio\python3\lib\socket.py", line 643, in socketpair
    ssock, _ = lsock.accept()
  File "C:\Users\shady\.platformio\python3\lib\socket.py", line 293, in accept
    fd, addr = self._accept()
socket.timeout: timed out

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

C:\Users\shady\.platformio\penv\lib\site-packages\platformio\__main__.py:131: RuntimeWarning: coroutine 'Server.serve' was never awaited
  exit_code = int(str(exc)) if str(exc).isdigit() else 1
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
Exception ignored in: <function BaseEventLoop.__del__ at 0x000002C2F6BE2790>
Traceback (most recent call last):
  File "C:\Users\shady\.platformio\python3\lib\asyncio\base_events.py", line 683, in __del__
    self.close()
  File "C:\Users\shady\.platformio\python3\lib\asyncio\proactor_events.py", line 684, in close
    self._close_self_pipe()
  File "C:\Users\shady\.platformio\python3\lib\asyncio\proactor_events.py", line 752, in _close_self_pipe
    self._ssock.close()
AttributeError: 'ProactorEventLoop' object has no attribute '_ssock'

	at c:\Users\shady\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10645
	at ChildProcess.o (c:\Users\shady\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1112:16)
	at ChildProcess._handle.onexit (node:internal/child_process:304:5)
  ```

ivankravets, 2023-06-01:
Please check your internet connection.

closed 2023-06-01 by ivankravets

ShadyEgypt, 2023-06-05:
My connection was good, I reinstalled vs code and platform io and everything worked for the first time, but when I try to work on it another day, I get the same error, for this reason, I tried installing platform io on vs code on ubuntu and now it works without problems there