platformio/platformio-vscode-ide #3722: vscode pio Startup error TLS CA certificate

BennyBread, 2023-05-23:
Hi, at first: 
I am using latest version of vscode 1.70.0 and PIO 3.11 running on Windows 10 22H2.

I already uninstalled PIO (latest version) vom vscode using extension menu. 
Afterwards I closed vscode completely.
Then I deleted my .platformio folder from my %userprofile%-folder
Then I reinstalled PIO inside of vscode extension manager.
(It recreated the .platformio folder and the files and folders in it do look ok to me.
The PIO ICON on the left side of vscore is there. When I click it it opens the side bar with the projects tasks and quick access view.)

NOW THE PROBLEM:
A click on PIO home open -> does not work 
create new project -> does not work

[image]
[vscode-app-1684830411950 create new project.log](https://github.com/platformio/platformio-vscode-ide/files/11541709/vscode-app-1684830411950.create.new.project.log)

I'm kinda stuck and don't know what else can I do..

BennyBread, 2023-05-23:
I managed to run pio inside of the terminal. but it gives the same error:

PS C:\Users\USER> pio home

  ___I_
 /\-_--\   PlatformIO Home       
/  \_-__\
|[]| [] |  http://127.0.0.1:8008/
|__|____|________________________

Open PlatformIO Home in your browser by this URL => http://127.0.0.1:8008/
Tool Manager: Installing platformio/contrib-piohome @ ~3.4.2
Error: Traceback (most recent call last):
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\USER\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__    
    return self.main(*args, **kwargs)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main        
    rv = self.invoke(ctx)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\cli.py", line 70, in invoke
    return super().invoke(ctx)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\USER\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\home\cli.py", line 92, in cli
    run_server(
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\home\run.py", line 65, in run_server
    contrib_dir = get_core_package_dir("contrib-piohome")
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\package\manager\core.py", line 46, in get_core_package_dir
    assert pm.install(spec)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 47, in install
    pkg = self._install(spec, skip_dependencies=skip_dependencies, force=force)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 99, in _install
    pkg = self.install_from_registry(
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\package\manager\_registry.py", line 29, in install_from_registry
    package = self.fetch_registry_package(spec)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\package\manager\_registry.py", line 93, in fetch_registry_package
    result = regclient.get_package(self.pkg_type, spec.owner, spec.name)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\registry\client.py", line 147, in get_package
    return self.fetch_json_data(
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\http.py", line 158, in fetch_json_data
    response = self.send_request(method, path, **kwargs)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\util.py", line 78, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\http.py", line 138, in send_request
    return getattr(self._session, method)(path, **kwargs)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\requests\sessions.py", line 602, in get
    return self.request("GET", url, **kwargs)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\platformio\http.py", line 61, in request
    return super().request(
  File "C:\Users\USER\.platformio\penv\lib\site-packages\requests\sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\requests\sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\requests\adapters.py", line 458, in send
    self.cert_verify(conn, request.url, verify, cert)
  File "C:\Users\USER\.platformio\penv\lib\site-packages\requests\adapters.py", line 261, in cert_verify
    raise OSError(
OSError: Could not find a suitable TLS CA certificate bundle, invalid path: .\Executable\C:\Programme (x86)\Microsoft SDKs\Azure\CLI2\Lib\site-packages\certifi\cacert.pem

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

PS C:\Users\USER>

ivankravets, 2023-05-23:
Is this path `.\Executable\C:\Programme (x86)\Microsoft SDKs\Azure\CLI2\Lib\site-packages\certifi\cacert.pem` valid?

BennyBread, 2023-05-23:
@ivankravets Yes, the path (and file) does exist. But I don't know what it's for.

Gungskraft, 2023-05-26:
I solved the same problem by installing pip-sytem-certs. 
C:\Users\<USER>\.platformio\penv\Scripts\python.exe -m pip install pip-sytem-certs
Or C:\Users\<USER>\.platformio\python3\python.exe -m pip install pip-sytem-certs