platformio/platformio-vscode-ide #3723: Inform developers that platformio.ini syntax is broken

Muzosh, 2023-05-23:
We should improve the UX and inform developers that config is broken even when developers did mistake. Thanks!

_Originally posted by @ivankravets in https://github.com/platformio/platformio-vscode-ide/issues/1398#issuecomment-1483218699_

rahji, 2023-05-23:
More information: I accidentally had the following line *twice* in platformio.ini:

```ini
board_build.f_cpu = 8000000L
```

That caused subsequent attempts to build, clean, or upload to fail with the error `['platformio-ide.build' not found]`

Removing the extra line caused it to work again.

closed 2023-06-02 by ivankravets

ivankravets, 2023-06-02:
Please re-test with the pre-release version. Navigate to the PlatformIO IDE extension in the VSCode and "Switch to the Pre-Release". Does it work now?
[image]

rahji, 2023-06-03:
Not only is the error more descriptive, I really appreciate the linting that happens in the platformio.ini editing window. I recreated the problem above (duplicate line) and it identified it successfully. Thanks!

commit f1bc6575 (2023-06-02): Introducing a powerful linting feature that highlights syntactical and stylistic issues in the "platformio.ini" configuration file // Resolve #3723