platformio/platformio-vscode-ide #3726: Can´t switch Project Environment

Emmpunkt, 2023-05-25:
Hello,
i can´t switch the Project Environment anymore.
This is for a while now and deinstalled and reinstalled VSCode and Platformio 3 times now, it doesn´t work.
If i click on the Button, nothing happens. Usually a window should pop out, but it just does nothing.
[image]
What can i do? It only switches to a the new environment if i make a new sketch...
Regards M.

ivankravets, 2023-06-01:
Please provide the contents of the `platformio.ini` file.

Emmpunkt, 2023-06-02:
It doesn´t matter which sketch i use....
but here the last one:
`[env:esp32dev]
platform = espressif32
board = esp32dev
upload_speed = 921600
framework = arduino
monitor_speed = 115200
lib_deps = 
	https://github.com/Savjee/EmonLib-esp32.git
	nrf24/RF24@^1.4.5
`

ivankravets, 2023-06-06:
Please re-test with the pre-release version. Navigate to the PlatformIO IDE extension in the VSCode and "Switch to the Pre-Release". Does it work now?
[image]

Emmpunkt, 2023-06-06:
No, did not work.
In the moment i´m working with platformio-ide.activateProjectOnTextEditorChange active, this works for me.
I deactivated it for the test ;)
There is still nothing happening if i click on "Switch PlatformIO Project Environment" Button.