platformio/platformio-vscode-ide #3731: JLink IP address setting from platformio.ini

OptifySudarshanPatil, 2023-05-31:
Hello All,
What is correct configuration for setting up JLink IP address to avoid manually entering in this box?
(command prompt methos is `jlink IP 192.168.0.1`)
[image]

84ace, 2023-06-01:
wow man, this was a real PITA to figure out.

Here is what i did:

in platformio.ini
`upload_protocol = custom
extra_scripts = extra_script.py`

in extra_script.py:
`from os import makedirs
from os.path import isdir, join
Import('env')

def _jlink_cmd_script(env, source):
    build_dir = env.subst("$BUILD_DIR")
    if not isdir(build_dir):
        makedirs(build_dir)
    script_path = join(build_dir, "upload.jlink")
    commands = ["h", "loadbin %s,0x0" % source, "r", "q"]
    with open(script_path, "w") as fp:
        fp.write("\n".join(commands))
    return script_path

env.Replace(
    __jlink_cmd_script=_jlink_cmd_script,
    UPLOADER="/Applications/SEGGER/JLink_V766a/JLinkExe",
    UPLOADERFLAGS=[
        "ip", "192.168.68.102",
        "-device", "NRF52840_XXAA",
        "-speed", "auto",
        "-if", "swd",
        "-autoconnect", "1"
    ],
    UPLOADCMD='"$UPLOADER" $UPLOADERFLAGS -CommanderScript ${__jlink_cmd_script(__env__, SOURCE)}'
)`

I hope that helps!
-Acea

OptifySudarshanPatil, 2023-06-01:
Hi @84ace 
Thanks for your replay!
My intention was to change jlink from "USB" to local network PC ip address where Jlink is connected to device. 
This extra_script.py have some strange error in my system 😅 

 debug_tool = custom
 debug_port = 192.168.1.101
 debug_server =
  $PLATFORMIO_CORE_DIR/packages/tool-jlink/JLink.exe
  -singlerun
  -if
  JTAG
  -select
  IP 192.168.1.101
  -port
  2331
  -device
  ATSAM3X8E

debug_init_cmds =
  define pio_reset_halt_target
      monitor reset
      monitor halt
  end
  define pio_reset_run_target
      monitor clrbp
      monitor reset
      monitor go
  end
  target extended-remote $DEBUG_PORT
  monitor clrbp
    monitor speed auto
    pio_reset_halt_target
    $LOAD_CMDS
    $INIT_BREAK