platformio/platformio-vscode-ide #3744: "Default" task of a project compiles another project

LupinIII, 2023-06-16:
If I select the "Default" task of a project and hit "Build" or "Upload" another project that is not active (non of its files are opened) gets built or uploaded.
If I select a specific task of a project (like "env:uno" in the example) build and upload works as expected.
It does not matter if I have a "default_envs = uno" in the project's (the one that should be compiled) platformio.ini.

This is not dependent on the current project. If I switch to another one it still executes the wrong tasks ("platformio-ide.activateProjectOnTextEditorChange" works for switching to the Default task of another project automatically, but does not change that fact that the wrong one gets executed).

[image]
[image]

LupinIII, 2023-06-16:
Hm, after a restart the problem is no longer there. I do not know what triggered it in the first place though. I have used the same workspace setup for several days in a row (not changing the projects included in the workspace). I did not create any new projects in that session, neither did I update any vscode or platformio extensions/libraries.
I was lucky that the "wrong" project was for a different microcontroller and programmer, otherwise I would have flashed my controller with a completely different firmware, that could have broken hardware.

MisiekDP, 2023-06-28:
I have 100% this same problem. And this is not first time when I have it. I can tell that I don't turn off the window too with VSCode so often and I noticed sometimes that I have updates too and problem still exist. For example I upload the video from today. 

https://github.com/platformio/platformio-vscode-ide/assets/22171619/7d6ae3e0-1ae6-4464-8d8a-0e2259fac749

d00m178, 2023-10-04:
the same issue. 
vscode v. 1.82.3
pio v. Core 6.1.11·Home 3.4.4

i've changed env in platformio.ini to different names, but it doesnt help. 
after restart vscode it start to work ok but when I open another project just for check the code and close it - it start to build this project although it is even not opened

closed 2025-04-26 by stale[bot]