platformio/platformio-vscode-ide #3748: Toolbar only shows a single icon

jobidon, 2023-06-22:
VS Code Version: 1.79.1, Release: 23164
OS: Linux x64 6.3.7-arch1-1
Pio Core 6.1.7, Home 3.4.4

The settings.json file has the following section:
```    ...
       "platformio-ide.toolbar": [
        {
            "text": "$(home)",
            "tooltip": "PlatformIO: Home",
            "commands": "platformio-ide.showHome"
        },
        {
            "text": "$(check)",
            "tooltip": "PlatformIO: Build",
            "commands": "platformio-ide.build"
        },
        {
            "text": "$(arrow-right)",
            "tooltip": "PlatformIO: Upload",
            "commands": "platformio-ide.upload"
        },
        {
            "text": "$(trashcan)",
            "tooltip": "PlatformIO: Clean",
            "commands": "platformio-ide.clean"
        },
        {
            "text": "$(beaker)",
            "tooltip": "PlatformIO: Test",
            "commands": "platformio-ide.test"
        },
        {
            "text": "$(plug)",
            "tooltip": "PlatformIO: Serial Monitor",
            "commands": "platformio-ide.serialMonitor"
        },
        {
            "text": "$(terminal)",
            "tooltip": "PlatformIO: New Terminal",
            "commands": "platformio-ide.newTerminal"
        }    ]
```
yet only the last item is actually displayed in the toolbar (the previous items do not appear at all in the toolbar).

I tried the following:
* hide/show the toolbar
* add/remove more elements in the toolbar
* hide/show other toolbar items (port switcher, etc)
* restart VS code
* reboot
* reset the toolbar settings
Neither of the steps above changes the behavior: only the last icon is ever shown in the toolbar as if there were only a single item in the toolbar array

If I change the last item to something else (modify the tooltip or the actual command), or if I change the order of the items in the array, then the toolbar is in fact updated accordingly, but still only the last item is displayed. What ever I try, only the last item of the array is ever displayed.

I noticed this problem after upgrading platform IO a few weeks ago, but haven't seen anyone else experiencing a similar issue.

ivankravets, 2023-06-22:
Please update VSCode and PlatformIO IDE extension to the latest versions.

closed 2023-06-22 by ivankravets

jobidon, 2023-06-23:
upgraded VS to: 
```
Version: 1.79.2
Release: 23166
Commit: 19c0f5cefaeb2637b45a9c7a356151fcfdcda1e9
Date: 2023-06-15T18:19:23.116Z
Electron: 22.3.10
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Linux x64 6.3.9-arch1-1
```
and using core 6.1.7-1 and Home 3.4.4. The problem persists and remains identical.
Is there a different version I should be upgrading to?

ivankravets, 2023-06-23:
https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide

Do you use the latest version?

jobidon, 2023-06-23:
ah, indeed that resolved the issue. Perhaps adding the IDE version number on the PlatformIO home page might help to eliminate confusion with version number? The page currently shows `Core 6.1.7·Home 3.4.4`, adding `IDE x.y.z` might help.
thanks for your input

ivankravets, 2023-06-23:
How was possible that you use outdated extension? Did you disable automatic updates?

jobidon, 2023-06-23:
I think VS Studio marketplace is not installed by default with the open source version of VS Studio, so I manually download the VSIX files and install them