platformio/platformio-vscode-ide #3756: Invalid simple block '' Couldn't figure out the error

darshandodamani, 2023-06-29:
Suddenly PlatformIO in VS code sstopped working. I am getting the error mentioned below:

Resolving esp32cam dependencies...
Platform Manager: Installing espressif32
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Platform Manager: espressif32@6.3.2 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch5 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20009.0
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-arduinoespressif32@3.20009.0 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40501.0
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-esptoolpy@1.40501.0 has been installed!
Removing unused dependencies...
Error: Traceback (most recent call last):
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\cli.py", line 70, in invoke
    return super().invoke(ctx)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\project\commands\init.py", line 104, in project_init_cmd
    install_project_dependencies(
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 106, in install_project_dependencies
    already_up_to_date = not install_project_env_dependencies(env, options)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_libraries(project_env, options),
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 204, in _install_project_env_libraries
    _uninstall_project_unused_libdeps(project_env, options)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 280, in _uninstall_project_unused_libdeps
    lm.uninstall(spec)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\package\manager\_uninstall.py", line 29, in uninstall
    return self._uninstall(spec, skip_dependencies)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\package\manager\_uninstall.py", line 34, in _uninstall
    pkg = self.get_package(spec)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 263, in get_package
    spec = self.ensure_spec(spec)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 133, in ensure_spec
    return spec if isinstance(spec, PackageSpec) else PackageSpec(spec)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 183, in __init__
    self._parse(self.raw)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 290, in _parse
    raw = parser(raw)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 315, in _parse_requirements
    self.requirements = tokens[1].strip()
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 230, in requirements
    else semantic_version.SimpleSpec(str(value))
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\semantic_version\base.py", line 647, in __init__
    self.clause = self._parse_to_clause(expression)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1043, in _parse_to_clause
    return cls.Parser.parse(expression)
  File "C:\Users\darsh\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1063, in parse
    raise ValueError("Invalid simple block %r" % block)
ValueError: Invalid simple block ''

ivankravets, 2023-07-05:
Try to remove `.pio` folder from the project and restart VSCode.

closed 2023-07-05 by ivankravets

DineshKazam, 2025-01-02:
> Try to remove `.pio` folder from the project and restart VSCode.
this worked. Thanks.

miloshev, 2025-04-24:
Thanks! This worked :)