platformio/platformio-vscode-ide #3761: Failed to upload to Teensy2 with two Teensy boards.

Raduq91, 2023-07-07:
I have two teensy boards connected to the pc, T4.0(as keyboard) and T2.0 .
When uploading with arduino ide it works to upload to T2, but when uploading with Visual Studio Code + Platformio it fails to upload and I have to replug T4 for it to work again.(unplugging T4, with only T2.0 it uploads)

```
Archiving .pio\build\teensy2\libFrameworkArduino.a
Indexing .pio\build\teensy2\libFrameworkArduino.a
Linking .pio\build\teensy2\firmware.elf
Checking size .pio\build\teensy2\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.1% (used 336 bytes from 2560 bytes)
Flash: [==        ]  24.8% (used 7986 bytes from 32256 bytes)
Building .pio\build\teensy2\firmware.hex
Configuring upload protocol...
AVAILABLE: teensy-cli, teensy-gui
CURRENT: upload_protocol = teensy-gui
Uploading .pio\build\teensy2\firmware.hex
Opening Teensy Loader...
Found 2 Teensy boards, but using auto-search to find board for upload.
Please use Tools > Ports(Teensy) to select the specific board.
============================================================ [SUCCESS] Took 3.92 seconds ============================================================ *  Terminal will be reused by tasks, press any key to close it. 
```
[image]

Arduino IDE uploading via Teensy native port option works without issue:

`0\hardware\teensy/../tools/teensy_post_compile -file=433mhz_2.ino -path=C:\Users\Radu\AppData\Local\Temp\arduino_build_905193 -tools=E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools -board=TEENSY2 -reboot -port=usb:0/140000/0/C -portlabel=hid#vid_16c0&pid_0478 Bootloader -portprotocol=Teensy`

Arduino IDE uploading via com10 port works without issue:

```
E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools/teensy_post_compile -file=433mhz_2.ino -path=C:\Users\Radu\AppData\Local\Temp\arduino_build_599706 -tools=E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools -board=TEENSY2 -reboot -port=COM10 -portlabel=COM10 -portprotocol=serial 
Teensy should be selected from "teensy ports" rather
than "Serial ports" in Arduino's Tools > Port menuFound 2 Teensy boards, but using auto-search to find board for upload.
Please use Tools > Ports(Teensy) to select the specific board.
```

When Arduino IDE is running teensy_ports.exe is running too. This does not happen with Visual Studio Code + Platformio.

Some info: https://www.pjrc.com/arduino-pluggable-discovery/

Raduq91, 2023-07-07:
platformio verbose upload:

```
avr-objcopy -O ihex -R .eeprom .pio\build\teensy2\firmware.elf .pio\build\teensy2\firmware.hex
<lambda>(["upload"], [".pio\build\teensy2\firmware.hex"])
AVAILABLE: teensy-cli, teensy-gui
CURRENT: upload_protocol = teensy-gui
teensy_post_compile -file=firmware -path="C:\Documente\documents\PlatformIO\Projects\433mhz_2\.pio\build\teensy2" -tools=C:\Users\Radu\.platformio\packages\tool-teensy -board=TEENSY2 -reboot
Opening Teensy Loader...
Found 2 Teensy boards, but using auto-search to find board for upload.
Please use Tools > Ports(Teensy) to select the specific board.
```

ivankravets, 2023-07-10:
Please forward to https://community.platformio.org/

closed 2023-07-10 by ivankravets