platformio/platformio-vscode-ide #3762: Unable to resolve configuration with compilerPath

cyberman54, 2023-07-09:
After starting VSCode while platformio extension loads, i get this error in output console:

`[7/8/2023, 3:05:35 PM] Unable to resolve configuration with compilerPath "C:/Users/xyz/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc.exe".  Using "C:\Program Files\LLVM\bin\clang.exe" instead.`

ivankravets, 2023-07-10:
Do you use the latest VSCofe and are all extensions updated?

cyberman54, 2023-07-12:
@ivankravets i think yes:
VScode v1.80.0 / OS Windows 10
Extensions: 
MS C/C++v1.16.3 
MS C/C++ Extension Pack v1.3.0

ivankravets, 2023-07-12:
Could you provide a simple project to reproduce this issue? You can attach it as ZIP to this ticket.

ivankravets, 2023-07-29:
Please reopen if you still need help

closed 2023-07-29 by ivankravets

cyberman54, 2023-07-29:
For others running in the same problem: root cause was the file `c_cpp_properties` found in .vscode subdirectory of the platformio project root dir. I deleted the whole project root dir and reinstalled it by cloning the project from github. Now there is no more .vscode subdirectory in the project root dir. Issue solved. I don't know, how this file this file was created before. At the op of the file it says it's autocreated (by platformio?).

/cc @ivankravets