platformio/platformio-vscode-ide #3765: PlatformIO: Can not find working Python 3.6+ Interpreter.

paranoid64, 2023-07-12:
Since the start of Visual studio code I get this message: PlatformIO Installer: Installing PlatformIO Core

I use the latest Linux Mint version:

```
whereis python3
python3: /usr/bin/python3 /usr/lib/python3 /etc/python3 /usr/share/python3 /usr/share/man/man1/python3.1.gz
```

```
python3 -V
Python 3.10.6
```

i don't have any other version of Python on it:

```
update-alternatives --config python
update-alternatives: Error: no alternatives for python
```

so now click: I have Python and enter  /usr/bin/python3  as full path.

and now i get:

**PlatformIO: Can not find working Python 3.6+ Interpreter.** 

But it works fine under VS Code Terminal:

```
~$ python
Python 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 

```
But I can't even stop that I could still work with it at all.

paranoid64, 2023-07-12:
unbelievable, i've been working on this problem all this time and then i write this issue and suddenly i find the solution: 

`sudo apt-get install python3-venv`

But why I have to install it all at once is a mystery to me.

closed 2023-07-12 by paranoid64

OliverColeman, 2024-09-01:
This got me too, on Ubuntu 22.04. Extremely uninformative error message!

MaxUint, 2024-10-08:
I also have this bug, but installing python3-venv doesn't fix it. I have python3 with conda insatlled on debian xcfe4, is conda breaking platformio? when I do a whereis python3, like the OP, I clearly have python installed, what is PIO's issue?