platformio/platformio-vscode-ide #3769: Cannot install platform io due to missing venv??

isFakeAccount, 2023-07-15:
```
Error: Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-venv

(MAY require administrator access `sudo`)
 -> Installer version: 1.2.0
Platform: Linux-6.2.6-76060206-generic-x86_64-with-glibc2.35
Python version: 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/kira/.platformio/penv

	at /home/kira/.vscode/extensions/platformio.platformio-ide-3.3.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5300
	at ChildProcess.i (/home/kira/.vscode/extensions/platfor...
```

I have venv. What venv it is talking about?

```
sudo apt install python3-venv

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-venv is already the newest version (3.10.6-1~22.04).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

isFakeAccount, 2023-07-15:
I fixed the issue by manually creating penv in `.platformio`

And in Vscode I set the `platformio-ide.customPATH` to `~/.platformio/penv/bin/`. Now I don't have to worry about vsc extension not being able to find venv for whatever reason.

luav, 2023-07-21:
I have had the same installation error when updating the PlatformIO IDE to v 3.3.0, which _does not occur when selecting previous versions (e.g., 3.2.0)_.
Thus, I believe that is a **v3.3.0-specifix error**:
```sh
Error: Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-venv

(MAY require administrator access `sudo`)
 -> Installer version: 1.2.0
Platform: Linux-5.15.0-76-generic-x86_64-with-glibc2.31
Python version: 3.11.4 (main, Jul 10 2023, 02:09:19) [GCC 9.4.0]
Python path: ~/.pyenv/versions/3.11.4/bin/python3
Creating a virtual environment at ~/.platformio/penv

    at ~/.vscode/extensions/platformio.platformio-ide-3.3.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5300
    at ChildProcess.i (....

```

schumar, 2023-07-25:
Not sure if this is the same issue, but I'm seeing sth similar:
```
$ python3.11 --version
Python 3.11.2
$ python3.11 get-platformio.py
Installer version: 1.2.0
Platform: Linux-6.1.0-9-amd64-x86_64-with-glibc2.36
Python version: 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0]
Python path: /usr/bin/python3.11
Creating a virtual environment at /home/martin/.platformio/penv
Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-venv

(MAY require administrator access `sudo`)
```

I tried running with `--verbose`, seeing

```
$ python3.11 get-platformio.py --verbose
Installer version: 1.2.0
Platform: Linux-6.1.0-9-amd64-x86_64-with-glibc2.36
Python version: 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0]
Python path: /usr/bin/python3.11
Creating a virtual environment at /home/martin/.platformio/penv
DEBUG: Current environment PATH /usr/local/bin:/local/stable-diffusion/pyenv/bin:/home/martin/bin:/home/martin/.local/bin:/usr/local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
DEBUG: Checking a Python candidate /usr/bin/python3.11
DEBUG: The Python 3.11.2 (/usr/bin/python3.11) interpreter is compatible.
DEBUG: Checking a Python candidate /usr/bin/python3.10
DEBUG: Error: The Python 3.10.9 (/usr/bin/python3.10) interpreter is not compatible.
Reason: Could not find Python `venv` module

Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-venv

(MAY require administrator access `sudo`)
```

and `strace`ing seems to confirm that the installer tries 3.11, which works, only then to try again with 3.10, which fails, and giving the error message.

```
45061 execve("/usr/bin/python3.11", ["/usr/bin/python3.11", "/usr/local/src/sidewinder-x2-marlin/tmp/get-platformio.py", "--no-shutdown-piohome", "check", "python"], [...]
45061 +++ exited with 0 +++
[...]
45062 execve("/usr/bin/python3.10", ["/usr/bin/python3.10", "/usr/local/src/sidewinder-x2-marlin/tmp/get-platformio.py", "--no-shutdown-piohome", "check", "python"], [...]
45062 +++ exited with 1 +++
```

Moving away `/usr/bin/python3.10` "fixed" this.

ivankravets, 2023-07-29:
Fixed in the latest PlatformIO IDE for VSCode. Please update

closed 2023-07-29 by ivankravets

raphaelyancey, 2023-08-03:
Still had the error. Downgrading the extension solved it.

ivankravets, 2023-08-04:
Have you read this solution https://github.com/platformio/platformio-core-installer/issues/85?