platformio/platformio-vscode-ide #3805: Inspect Memory tools exits with "Invalid params"

trullock, 2023-08-15:
PIO Home -> New Project

Platformio.ini
```
[env:esp32s3box]
platform = espressif32
board = esp32s3box
; build_type = debug
framework = arduino
```

With or without debug build mode

Environment -> Build

PIO Home -> Inspect

Uncheck Code (although this makes no difference)

Inspect

Progress bar runs to 90%

It stops and shows: "Error: Invalid params"

Halp

darthmelak, 2023-08-17:
The same happens with a fresh generated ESP8266 project  

ini:
```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
```
Same uninformative "Invalid params" error message.

siglun88, 2023-08-21:
Same error saying "Invalid params" when running project memory inspection, on freshly generated project using NodeMCU-32s and esp-idf.

```
[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
framework = espidf
```

amal097acl, 2023-08-21:
Same Error with STM32F103C8

Only `Inspect Memory` gives error. `Check Code` runs without any issues
```
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
```

benmalenfant, 2023-08-22:
Same issue here with stm32f103c8, issue only happens with Inspect Memory check runs fine

strnk, 2023-08-23:
I had the same problem with PIO core 6.1.10 (ESP8266 platform / D1 mini board).
The latest dev branch 6.1.11a2 does not have this problem: `pio upgrade --dev` in the PlatformIO CLI, then restart the IDE.

MikahB, 2023-08-27:
Same problem here with attiny48.  Tried running `pio upgrade --dev` from CLI but it errors out because pio.exe is in use (by Windows) and then I have to remove/reinstall the extension.

0x0fe, 2023-09-04:
same error, invalid params with IDF 5.0.2 brand new project, nothing in the terminal.

[image]

and calling `pio upgrade --dev` results in the following error 

[image]

trullock, 2023-12-16:
This is now working for me in PIO 6.1.11

ivankravets, 2024-01-26:
> This is now working for me in PIO 6.1.11.

yes, it was fixed in the PIO Core. Sorry for the issue.

closed 2024-01-26 by ivankravets

grisharav, 2025-02-23:
@ivankravets - this happens to me as well, latest available PIO (6.1.17). No info just `Invalid params` in Errors. Happens in debug/release builds