platformio/platformio-vscode-ide #3809: "Upload & Monitor" Failing but "Upload" then "Monitor" works

Jordan-Loeser, 2023-08-18:
Hello! I just installed the PlatformIO IDE to use for an Adafruit MatrixPortal M4 project. I am able to run the "Build", "Upload", and "Monitor" commands successfully, but when I run "Upload and Monitor" I get the following error:
```shell
...
[============================  ] 95% (47/49 pages)
[============================= ] 97% (48/49 pages)
[==============================] 100% (49/49 pages)
Verify successful
Done in 0.871 seconds
UserSideException: [Errno 2] could not open port /dev/cu.usbmodem101: [Errno 2] No such file or directory: '/dev/cu.usbmodem101'

 *  The terminal process "platformio 'run', '--target', 'upload', '--target', 'monitor', '--environment', 'adafruit_matrix_portal_m4', '--upload-port', '/dev/cu.usbmodem101', '--monitor-port', '/dev/cu.usbmodem101'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

It seems to think that the port is already occupied even though I have no serial monitors running. I have tried setting `"platformio-ide.autoCloseSerialMonitor": true` and `"platformio-ide.reopenSerialMonitorDelay": 2000` with no luck. I have also disabled all other VSCode extensions.

I am using:
* VSCode `v1.81.1`
* PlatformIO IDE `v3.3.1`

and the following `platformio.ini`
```ini
[env:adafruit_matrix_portal_m4]
platform = atmelsam
board = adafruit_matrix_portal_m4
framework = arduino
monitor_speed = 9600
lib_deps = 
    ; Adafruit Protomatter
    adafruit/Adafruit Protomatter@^1.5.10
```

May be related to previous closed issue #3656

Jordan-Loeser, 2023-08-18:
I have also tried running this in a regular terminal using the following command with the same error:
```shell
~/.platformio/penv/bin/platformio run --target upload --target monitor --environment adafruit_matrix_portal_m4 --upload-port /dev/cu.usbmodem101 --monitor-port /dev/cu.usbmodem101
```

crb123456, 2023-08-19:
I am getting a somewhat different error but similar.  Mine says
Could not open comXX the port does not exist
PIO devices command shows the port.  Reboot does nothing.  Unless you have another program using the port, I believe the PIO monitor still using it 

In the monitor window (mine is at bottom)  on right hand side, use trash icon to delete all current PIO monitors. 
This leaves you with a power shell monitor.  Kill that with the trashcan.  This will close the monitor window.
In PIO tools click upload again and PIO should try to connect.

Jordan-Loeser, 2023-08-20:
It looks like this is due to the fact there is not enough delay between the board being reset and the USB port being detected. I was able to work around this by following this [community answer](https://community.platformio.org/t/serial-monitor-port-not-automatically-selected-after-upload/11922/4). It required two steps:

* Creating an `extra_script` that manually handled a 2000ms delay between upload and monitor
* Adding a `delay(2000)` to my `setup()` so that I didn't miss any important output

I just played around with some different values to find that 2000ms was the lowest I could go without getting an error.

Closing this as resolved, but it would be great to have a way to do this out of the box with PlatformIO or to flag it in the documentation.

closed 2023-08-20 by Jordan-Loeser