platformio/platformio-vscode-ide #3859: Start PIO Home Server

techanshuman, 2023-10-07:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.83.0
  PIO IDE: v2.2.1
  System: Linux, 5.4.0-150-generic, x64

  %23 Exception
  ```
  Error: AttributeError: Traceback (most recent call last):
  File "/home/an/.platformio/penv/lib/python3.6/site-packages/platformio/__main__.py", line 103, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/home/an/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/home/an/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/home/an/.platformio/penv/lib/python3.6/site-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
  File "/home/an/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/an/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/an/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/an/.platformio/penv/lib/python3.6/site-packages/platformio/home/cli.py", line 97, in cli
    home_url=home_url,
  File "/home/an/.platformio/penv/lib/python3.6/site-packages/platformio/home/run.py", line 102, in run_server
    log_level="warning",
  File "/home/an/.platformio/penv/lib/python3.6/site-packages/uvicorn/main.py", line 461, in run
    server.run()
  File "/home/an/.platformio/penv/lib/python3.6/site-packages/uvicorn/server.py", line 67, in run
    return asyncio.run(self.serve(sockets=sockets))
AttributeError: module 'asyncio' has no attribute 'run'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /home/an/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:5625
	at ChildProcess.c (/home/an/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (node:events:513:28)
	at maybeClose (node:internal/child_process:1091:16)
	at Socket.<anonymous> (node:internal/child_process:449:11)
	at Socket.emit (node:events:513:28)
	at Pipe.<anonymous> (node:net:322:12)
  ```

techanshuman, 2023-12-15:
issue is still not resolved

ivankravets, 2024-01-06:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3924

closed 2024-01-06 by ivankravets