platformio/platformio-vscode-ide #3860: Start PIO Home Server

kasir86, 2023-10-08:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.83.0
  PIO IDE: v3.3.1
  System: Windows_NT, 10.0.22621, x64

  %23 Exception
  ```
  Error: InvalidJSONFile: Could not load broken JSON: C:\Users\kasir\.platformio\packages\contrib-pioremote\.piopm

	at c:\Users\kasir\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10851
	at ChildProcess.c (c:\Users\kasir\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4598)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1091:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

kasir86, 2023-10-08:
1

ivankravets, 2024-01-06:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3786

closed 2024-01-06 by ivankravets