platformio/platformio-vscode-ide #3871: Failed to locate python interpreter

va1labh, 2023-10-20:
I'm trying to install the plugin in VS Code on Ubuntu 22.04 LTS. 

I have python 3.10.6 installed at `usr/bin/python3`'; I did specify the same path, but it failed to identify!

ZoyKad, 2023-10-26:
Your python interpreter must have the "venv" module. You can get it by installing another version of python such as:
```bash
sudo apt install python3.11-venv
```
Then restart VSC, retry to install platformio core and when it says you it does not find the python3 interpreter, click "I have python installed" and specify the path : `/usr/bin/python3.11`.

It should work then.

ivankravets, 2024-01-26:
See https://github.com/platformio/platformio-core-installer/issues/85

closed 2024-01-26 by ivankravets