platformio/platformio-vscode-ide #3884: .pio folder not generated on startup

JL-Money, 2023-11-04:
Since I've installed PlatformIO last December, it has been working well. But today when I tried to build my code like I have done so many times before, it errored. The last time I have used PlatformIO before today is about 1 month ago.
So, I attempted to fix the issue by uninstalling PlatformIO from VSCode, deleting the C:/Users/<user>/.platformio folder, reinstalling PlatformIO, and updating it with `python -m pip install -U platformio`. Now, it's throwing the error that the .pio folder isn't found.
I did notice that for a PIO project to work, both a .vscode and .pio folder are needed inside the project directory. They should both be auto-generated, but my .pio file is not being generated.

**Configuration**
VSCode: 1.84.0
PIO IDE: 3.3.1
System: Windows 10, Version 22H2, x64

**Exception**
```
FileNotFoundError: Traceback (most recent call last):
  File "C:\Users\josep\.platformio\python3\Lib\pathlib.py", line 1116, in mkdir
    os.mkdir(self, mode)
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\josep\\Documents\\GitHub\\Excalibur-FW\\.pio\\libdeps\\steering2'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\josep\.platformio\python3\Lib\pathlib.py", line 1116, in mkdir
    os.mkdir(self, mode)
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\josep\\Documents\\GitHub\\Excalibur-FW\\.pio\\libdeps'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\josep\.platformio\python3\Lib\pathlib.py", line 1116, in mkdir
    os.mkdir(self, mode)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'C:\\Users\\josep\\Documents\\GitHub\\Excalibur-FW\\.pio'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\platformio\run\cli.py", line 145, in cli
    process_env(
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\platformio\run\cli.py", line 208, in process_env
    ).process()
      ^^^^^^^^^
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\platformio\run\processor.py", line 81, in process
    install_project_env_dependencies(
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 133, in install_project_env_dependencies
    _install_project_env_libraries(project_env, options),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 207, in _install_project_env_libraries
    _uninstall_project_unused_libdeps(project_env, options)
  File "C:\Users\josep\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 287, in _uninstall_project_unused_libdeps
    storage_dir.mkdir(parents=True)
  File "C:\Users\josep\.platformio\python3\Lib\pathlib.py", line 1120, in mkdir
    self.parent.mkdir(parents=True, exist_ok=True)
  File "C:\Users\josep\.platformio\python3\Lib\pathlib.py", line 1120, in mkdir
    self.parent.mkdir(parents=True, exist_ok=True)
  File "C:\Users\josep\.platformio\python3\Lib\pathlib.py", line 1121, in mkdir
    self.mkdir(mode, parents=False, exist_ok=exist_ok)
  File "C:\Users\josep\.platformio\python3\Lib\pathlib.py", line 1116, in mkdir
    os.mkdir(self, mode)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'C:\\Users\\josep\\Documents\\GitHub\\Excalibur-FW\\.pio'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

JL-Money, 2023-12-24:
I am still having this issue.

ivankravets, 2024-01-26:
Do you use any espionage software (3rd party antivirus tools)? They can block PlatformIO.

closed 2024-01-26 by ivankravets