platformio/platformio-vscode-ide #3913: Start PIO Home Server

GergoGyoriIGrain, 2023-12-05:
During setting up a roject my computer freeze, and after restartI tried to start a new project. This is the log. 

I tried to uninstall and install the extension, getting the same message 

%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.84.2
  PIO IDE: v3.3.1
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: InvalidJSONFile: Could not load broken JSON: C:\Users\share\.platformio\packages\framework-arduino-avr\.piopm

	at c:\Users\share\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10851
	at ChildProcess.c (c:\Users\share\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4598)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1091:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

GergoGyoriIGrain, 2023-12-05:
solved, had to delete manually the files from C:\Users\share\.platformio\ since files are still there even I uninstall the extension

ivankravets, 2024-01-06:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3786

closed 2024-01-06 by ivankravets