platformio/platformio-vscode-ide #3947: Start PIO Home Server

78lobo78, 2024-01-26:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.85.2
  PIO IDE: v3.3.3
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: C:\Users\escor.DESKTOP-APC7SEB\.platformio\penv\Scripts\python.exe: No module named platformio

	at c:\Users\escor.DESKTOP-APC7SEB\.vscode\extensions\platformio.platformio-ide-3.3.3-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:11001
	at ChildProcess.i (c:\Users\escor.DESKTOP-APC7SEB\.vscode\extensions\platformio.platformio-ide-3.3.3-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4621)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1091:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

ivankravets, 2024-01-28:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3933

closed 2024-01-28 by ivankravets

78lobo78, 2024-01-28:
> <platformio/platformio-vscode-ide/issue/3947/issue_event/11620515495@
> github.com>

Hello, I don't know what happens that it doesn't install it for me. Why can
it be due?

ivankravets, 2024-01-28:
Do you use espionage software (3rd party antivirus tools, russian software)? They remove PlatformIO installation.

78lobo78, 2024-01-28:
If I use Malwarebytes and AVG Internet Security

ivankravets, 2024-01-28:
Exclude this folder `C:\Users\escor.DESKTOP-APC7SEB\.platformio` or disable it for a while.

78lobo78, 2024-01-28:
Removing and excluding it doesn't install it for me. I don't know what else
to do.