platformio/platformio-vscode-ide #3948: Start PIO Home Server

itech7645738, 2024-01-27:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.85.2
  PIO IDE: v3.3.3
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: C:\Users\itech robotics\.platformio\penv\Scripts\python.exe: No module named platformio

	at c:\Users\itech robotics\.vscode\extensions\platformio.platformio-ide-3.3.3-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:11001
	at ChildProcess.i (c:\Users\itech robotics\.vscode\extensions\platformio.platformio-ide-3.3.3-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4621)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1091:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

itech7645738, 2024-01-27:
can you help me please

ivankravets, 2024-01-28:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3933

closed 2024-01-28 by ivankravets