platformio/platformio-vscode-ide #3949: Upgrade to v3.3.0 installs wrong Python version on Apple ARM

Jason2866, 2024-01-27:
@ivankravets 
Upgrading installs a non native Python version on Apple Silicon. Instead of an Apple ARM python version a Apple x86 python version is installed. **This brakes any use of native Apple ARM tools and toolchains.**

Please provide a native Python version for Apple ARM.

ivankravets, 2024-01-27:
A temporary solution is to disable [platformio-ide.useBuiltinPython](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-usebuiltinpython) setting. Remove `~/.platformio/penv` and `~/.platformio/python3` folders. Restart VSCode.

Jason2866, 2024-01-27:
Just did this already to solve ;-)

e7h4n, 2024-03-20:
I got the [same error](https://github.com/lvgl/lv_platformio/pull/63), it really hard to find the root cause :(

Jason2866, 2024-03-20:
@ivankravets Any news, this is not a minor bug. It brakes every clean fresh installation on Apple ARM. 
Building an portable Python seems not so hard. Tried this with Github Actions.
https://github.com/Jason2866/python-portable-darwin_arm

Jason2866, 2024-03-20:
@ivankravets Now for both architectures
https://github.com/Jason2866/python-portable-darwin_arm/releases

ivankravets, 2024-03-20:
Could you remove the following folders?
- `~/.platformio/.cache`
- `~/.platformio/penv`
- `~/.platformio/python3`

Please ensure that `platformio-ide.useBuiltinPIOCore` and `platformio-ide.useBuiltinPython` options are enabled in the PlatformIO IDE for VSCode.

Does it work now?

Jason2866, 2024-03-20:
Yes, it does work now!

closed 2024-03-20 by Jason2866

ivankravets, 2024-03-20:
Thanks a lot for the idea with the GH actions 🙏

Jason2866, 2024-03-22:
@ivankravets  Stumbled over this official Github repo https://github.com/actions/python-versions
Maybe you can use in future, avoiding double work?

ivankravets, 2024-03-23:
Their packages are OS-dependent and require superuser access. Indeed, we use official Windows packages from python.org and the only macOS packages are cross-compiled to portable format.