platformio/platformio-vscode-ide #3950: Why does GCC compiler using ARM64 format on Apple silicon compile X86_64 format?

CNSTJiatongLee, 2024-01-27:
I set platform=native on Mac M1 and it compiles using my native gcc compiler. Why is my native gcc compiler in ARM64 format and able to compile x86_64 files? This caused an error when linking to sdl2 in arm64 format that homebrew downloaded
[image]
[image]
[image]

Jason2866, 2024-01-28:
see #3949

ivankravets, 2024-01-28:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3949

closed 2024-01-28 by ivankravets