platformio/platformio-vscode-ide #3968: Error: Unknown board ID 'd1_wroom_02'

bLind17, 2024-02-06:
Hi there!

I recently got my hands on a few WeMos D1 Esp-Wroom-02 devices. I use vscode and PlatformIO to program them.

PlatformIO IDE extension is v3.3.3
vscode is up to date

I used the following .ini when trying to flash:
```ini
[env:d1_wroom_02]
platform = espressif8266
board = d1_wroom_02
framework = arduino
monitor_speed = 115200
```

And I have successfully used this to flash two days ago on another PC.

```
Resolving d1_wroom_02 dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing d1_wroom_02 (platform: espressif8266; board: d1_wroom_02; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Error: Unknown board ID 'd1_wroom_02'
========================== [FAILED] Took 1.20 seconds ==========================
```

Does anyone have an idea what's going on here?
The board is in the registry: https://docs.platformio.org/en/latest/boards/espressif8266/d1_wroom_02.html

Thanks!

jankoscharf, 2024-02-06:
I'm just on Platformio Core 6.1.14a1·Home 3.4.4 (on VSCodium, but this should i.m.h.o. not make any difference)
Maybe an update would help?

bLind17, 2024-02-08:
Thanks for the reply. I don't know how to check the core version, but I got it to work in another way:

In vscode on `PIO home` I went to the `Platforms` tab where `Espressif 8266` is installed. There was an update available, I ran it and then it worked...

```
Processing d1_mini_lite (platform: espressif8266; board: d1_mini_lite; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/framework-arduinoespressif8266 @ ~3.30102.0
Downloading  [#######################-------------]   66%  00:00:04
```

closed 2024-02-08 by bLind17