platformio/platformio-vscode-ide #3982: Cannot read properties of undefined (reading ‘id’)

alanmasu, 2024-02-17:
Hi,
I’m using PlatformIO for developing a project whit an RP2040 and an ESP01 connected to it.

So I’ve created multiple environments for differentiating sources, such functionality testing files and real application sources, but also for sources for the ESP01 and the RP2040:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
extra_configs = upload_port.ini
default_envs = pico

[env]
extra_scripts = pre:git_script.py

[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.mcu = rp2040
board_build.core = earlephilhower
monitor_speed = 115200
build_src_filter = -<ESP>

[env:test_WebServer]
extends = env:pico
build_src_filter =  -<./main.cpp>
                    +<./common>
                    +<../testing/test_WebServer>
lib_deps =  https://github.com/adafruit/Adafruit-GFX-Library.git#126007f
            https://github.com/adafruit/Adafruit_SSD1306.git#7a4d33e
            https://github.com/adafruit/Adafruit_NeoPixel.git@1.12.0

[env:esp]
framework = arduino
monitor_speed = 115200
board_build.filesystem = littlefs
build_src_filter =  -<.>
                    +<ESP>
lib_deps = https://github.com/cotestatnt/esp-fs-webserver.git#1.2.7

[env:esp01]
extends = env:esp
platform = espressif8266@4.2.1
board = esp01

[env:esp32]
extends = env:esp
platform = espressif32@6.4.0
board = esp32dev
```
and it’s ok, but when I change the environment buttons in the bottom navigation tab:
[image]

shows the `Cannot read properties of undefined (reading 'id')` error and for upload/build/etc… I have to use the PlatfomIO tab on the nav bar instead:
[image]

In particular this error is triggered when environment is changed and also after a pair of hours of work! 
There is a discussion opened also in the Community Forum reachable [here](https://community.platformio.org/t/cannot-read-properties-of-undefined-reading-id/38524/1)

I’m working on Windows 10 and VSCode 1.86.1 and PIO (core: 6.1.14) (Home: 3.4.4)
There is any solution?? This is very uncomfortable…

Thanks in advance!
Alan

Strooom, 2024-02-17:
I’m having this behavior as well, already for several months. Here is what’s happening:

* after switching the [env] there is some activity from PlatformIO extension, showing a clock badge on the PlatformIO icon in the left-side taskbar.

* if you try to build before this activity is completed, you get the Cannot read properties of undefined (reading ‘id’) error. That’s ok, it’s busy.

* after some time working (typically a few hours), even if you wait until the environment switching activity is completed, this error also pops up. It only pops up when using the bottom toolbar. When using the left-side bar (Project Tasks), it still works. The problem is not only for build, also for test.

* when you exit VSCode and re-open it, the bottom toolbar works again.

So currently my workaround is to close VSCode and open it again, but still it is an annoying thing.

My setup is Win11 and all up-to-date with PlatformIO.

My repo is this : Strooom/MuMo-v2-Node-SW: Firmware for the Museum Monitor V2 node (github.com) in case you want to check my platformio.ini

I'll try to find a way to reproduce it in a deterministic way. Will comment here if I find one.

Will also ask my colleague working on MacOs if he ever experienced this. (Working on the same repository)

mattia424, 2024-02-24:
Hi, I have this problem too.
I have created two envs which differ only by a build_flag and upload_port. 

When I use env default it compiles and uploads to both tabs while if I move to env specific I get the toast "Cannot read properties of undefined (reading 'id')"

Platformio version Core 6.1.11-Home 3.4.4

I don't know what other info I could add to help reproduce the problem. 

Mattia

QuinveY, 2024-03-17:
Exactly the same problem. 
Using PIO Core: 6.1.10 & PIO Home: 3.4.4

When switching the PIO Project Environment the buttons on the bottom do not work anymore, but on the left side unther project tasks they work fine.

andyjbm, 2024-03-26:
+1 Here.
Thought I was going mad trying to compile. Happened just after adding another environment. Could no longer use the buttons on the toolbar - getting the error as in the title.

I'm using a dozen or so environments with build flags to define a macro for different code compiling #ifdefs
The environment I added was the same as the dozen or so others I have except a build flag -D was changed:

`
[env:ESP32_s2_mini_NoMsgs]
extends = env:Release32_s2_mini
build_flags = 
	${env:Release32_s2_mini.build_flags}
	-D GENIE_TYPE_NOMSG=true
`

Ended up here after googling, Hadn't thought to try the Project Tasks list menu and hey presto they work, but the toolbar buttons don't.

My addition. Shortcut keys (for the toolbar buttons) don't work either, however the notification message has a warning triangle whereas pressing a toolbar button gets a notification with a red X. The message is exactly the same.

Andy.

andyjbm, 2024-03-26:
Hi Quick addition,

May not be relevant but the more info and all that...

I had a second vscode/platformio instance open with another project. That toolbar still worked while the broken one didn't.

A.

pdurys, 2024-05-27:
Same here.
I am using latest [T-Display-S3](https://github.com/Xinyuan-LilyGO/T-Display-S3) examples.
Some envs throws that error right away, some works and at the same time side menu works as expected. 
Only taskbar throws that error.
Interestingly I do not have that problem with projects which have only one env in platformio.ini

rpolitex, 2024-08-20:
**POSSIBLE TEMPORARY SOLUTION**

I have such a problem, when the CMake Extension ([v.0.0.17](https://marketplace.visualstudio.com/items?itemName=twxs.cmake), id: `twxs.cmake`) for VSCode is enabled.

Disabling of Extention solves the problem.

Strooom, 2024-08-20:
Ok, thanks for that hint. I have CMake also enabled. Will disable it and keep an eye if this solves it for me.

olicooper, 2024-10-10:
Same issue here on pio 6.1.6

I usually notice it when rebuilding the project after editing an environment name or adding a new environment in the `platformio.ini` file. The only fix is to restart VSCode (and lose all my edit history 😢) but I bet if we could just restart the extension that would also work.

mvdbastos, 2024-11-29:
I had the same problem, for me the solution was to change [env] for [common] takin this example from PlatformIO's website https://docs.platformio.org/en/latest/projectconf/index.html

Example:
```
...
[platformio]
extra_configs = upload_port.ini
default_envs = pico

[common]
extra_scripts = pre:git_script.py

[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
...
```

GerhardMalan, 2025-04-21:
This issue also happened for me today. Created a third [env] section in theplatformio.ini, identical to the first two, except for a different partition file. After selecting the new environment, when I hit build or upload in the bottom toolbar I get "_Cannot read properties of undefined (reading 'id')_". From the sidebar menu I can build/upload without any issue. If I select either of the first two environments, the bottom toolbar menu works fine without any errors.

GerhardMalan, 2025-04-21:
Just realized the same issue applies to the 'Serial Monitor' button in the bottom toolbar. As above, the corresponding side menu item has no issue.

closed 2025-04-29 by stale[bot]