platformio/platformio-vscode-ide #3990: Request for "Automatic Function Prototypes" Feature in PlatformIO

buteomont, 2024-02-24:
I'm a regular user of PlatformIO and have found it to be an excellent tool for embedded development. However, I believe there's an opportunity to enhance the development experience further by introducing a feature for **automatic generation of function prototypes**.

Currently, developers often need to manually write function prototypes in header files or at the beginning of source files to ensure proper code organization and compilation. This process can be time-consuming and error-prone, especially in projects with many functions or frequent changes to function signatures.

Therefore, I would like to propose the addition of an "Automatic Function Prototypes" feature to PlatformIO. This feature would analyze the source code within a project and generate function prototypes automatically, either at the top of each source file or in separate header files. This would streamline the development process, reduce the likelihood of errors, and improve overall productivity.

I believe that implementing this feature would make PlatformIO an even more attractive choice for embedded development projects and would greatly benefit the PlatformIO community.

Thank you for considering this suggestion. I look forward to hearing your thoughts on this proposal and to potentially seeing this feature implemented in a future release of PlatformIO.

Best regards,
Buteomont

rtek1000, 2024-03-20:
There is an extension for C, but I couldn't make it work in the Arduino UNO Platformio code C++. Maybe it works with ESP-IDF (C) for ESP32.

> The C Auto Prototypes extension generates prototypes for all functions in your program at the press of a button! Well, at the press of three buttons. Simply press the shortcut Alt + Shift + Q and the extension will scan all the functions defined after main() and add their prototypes above main(). It’s important to note that it will do this for functions defined after the main function! But that’s not all: If you ever change one of the functions, simply press Alt + Shift + Q again and C Auto Prototypes will update the existing prototypes to reflect the changes. It’s simple, quick and very efficient.

https://www.ohsat.com/tutorial/mdmisc/c-auto-prototype-extension/

rtek1000, 2024-03-20:
I made a basic script to generate the prototypes, it is available at:

https://github.com/rtek1000/CPlusPlus_Prototype_Generator/tree/main/Python

nbourre, 2024-10-17:
Is that a PlatformIO thing? I noticed that I needed to add function prototypes in my main file which wasn't required in Arduino IDE.

closed 2025-05-08 by stale[bot]