platformio/platformio-vscode-ide #3997: Build process fails randomly with ever-changing error messages

fredlcore, 2024-03-01:
I have a project (https://github.com/fredlcore/BSB-LAN) that I use with PlatformIO on MacOS Sonoma 14.2.1. For a while nowmore and more often I run into errors in the build process that sometimes go away just by clicking "Build" again, sometimes not. Sometimes it helps for a few builds if I rename the project (for example from `[env:esp32-JoyIt]` to `[env:esp32-Joy-It]`), sometimes not. Also, the same code will compile fine for esp32-evb, but not for an esp32doit-devkit-v1 - or vice versa.
I have tried "Full Clean" several times and have also removed the corresponding platforms from `.platformio/platforms` so that they got re-downloaded and reinstalled, as well as removing the build and libdeps folders from the .pio folder. Nothing helped.

These are some of the errors that I'm randomly getting:
```
Building in release mode
[src/BSB_LAN/BSB_LAN.ino.cpp] /Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/.pio/build/esp32-JoyIt/.sconsign311.dblite: No such file or directory
```

```
Building in release mode
Compiling .pio/build/esp32-JoyIt/src/BSB_LAN.ino.cpp.o
Compiling .pio/build/esp32-JoyIt/src/src/ArduinoMDNS/MDNS.cpp.o
Compiling .pio/build/esp32-JoyIt/src/src/ArduinoMDNS/utility/EthernetUtil.c.o
Compiling .pio/build/esp32-JoyIt/src/src/BSB/BSBSoftwareSerial.cpp.o
Compiling .pio/build/esp32-JoyIt/src/src/BSB/bsb.cpp.o
Compiling .pio/build/esp32-JoyIt/src/src/Base64/src/Base64.cpp.o
Compiling .pio/build/esp32-JoyIt/src/src/BlueDot_BME280/BlueDot_BME280.cpp.o
Compiling .pio/build/esp32-JoyIt/src/src/CRC32/CRC32.cpp.o
Compiling .pio/build/esp32-JoyIt/src/src/DHTesp/DHTesp.cpp.o
Compiling .pio/build/esp32-JoyIt/src/src/DallasTemperature/DallasTemperature.cpp.o
Compiling .pio/build/esp32-JoyIt/src/src/OneWireNg/OneWireNg.cpp.o
Compiling .pio/build/esp32-JoyIt/src/src/OneWireNg/OneWireNg_BitBang.cpp.o
src/BSB_LAN/src/Base64/src/Base64.cpp:143:19: fatal error: opening dependency file .pio/build/esp32-JoyIt/src/src/Base64/src/Base64.cpp.d: No such file or directory
 Base64Class Base64;
                   ^
compilation terminated.
*** [.pio/build/esp32-JoyIt/src/src/Base64/src/Base64.cpp.o] Error 1
src/BSB_LAN/src/BSB/bsb.cpp:685:1: fatal error: opening dependency file .pio/build/esp32-JoyIt/src/src/BSB/bsb.cpp.d: No such file or directory
 }
 ^
compilation terminated.
*** [.pio/build/esp32-JoyIt/src/src/BSB/bsb.cpp.o] Error 1
src/BSB_LAN/src/CRC32/CRC32.cpp:56:1: fatal error: opening dependency file .pio/build/esp32-JoyIt/src/src/CRC32/CRC32.cpp.d: No such file or directory
 }
 ^
compilation terminated.
*** [.pio/build/esp32-JoyIt/src/src/CRC32/CRC32.cpp.o] Error 1
src/BSB_LAN/src/BlueDot_BME280/BlueDot_BME280.cpp:455:1: fatal error: opening dependency file .pio/build/esp32-JoyIt/src/src/BlueDot_BME280/BlueDot_BME280.cpp.d: No such file or directory
 }
 ^
compilation terminated.
*** [.pio/build/esp32-JoyIt/src/src/BlueDot_BME280/BlueDot_BME280.cpp.o] Error 1
src/BSB_LAN/src/DHTesp/DHTesp.cpp:486:1: fatal error: opening dependency file .pio/build/esp32-JoyIt/src/src/DHTesp/DHTesp.cpp.d: No such file or directory
 }
 ^
compilation terminated.
*** [.pio/build/esp32-JoyIt/src/src/DHTesp/DHTesp.cpp.o] Error 1
src/BSB_LAN/src/DallasTemperature/DallasTemperature.cpp:603:1: fatal error: opening dependency file .pio/build/esp32-JoyIt/src/src/DallasTemperature/DallasTemperature.cpp.d: No such file or directory
 }
 ^
compilation terminated.
*** [.pio/build/esp32-JoyIt/src/src/DallasTemperature/DallasTemperature.cpp.o] Error 1
src/BSB_LAN/src/OneWireNg/OneWireNg_BitBang.cpp:219:1: fatal error: opening dependency file .pio/build/esp32-JoyIt/src/src/OneWireNg/OneWireNg_BitBang.cpp.d: No such file or directory
 }
 ^
compilation terminated.
*** [.pio/build/esp32-JoyIt/src/src/OneWireNg/OneWireNg_BitBang.cpp.o] Error 1
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_LAN/BSB_LAN.ino:8625:1: fatal error: opening dependency file .pio/build/esp32-JoyIt/src/BSB_LAN.ino.cpp.d: No such file or directory
 }
 ^
compilation terminated.
*** [.pio/build/esp32-JoyIt/src/BSB_LAN.ino.cpp.o] Error 1
FileNotFoundError: [Errno 2] No such file or directory: '/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/.pio/build/esp32-JoyIt/.sconsign311.tmp':
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/Main.py", line 1470:
    _exec_main(parser, values)
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/Main.py", line 1431:
    _main(parser)
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/Main.py", line 1174:
    nodes = _build_targets(fs, options, targets, target_top)
[... 83 lines cut ...]
    nodes = _build_targets(fs, options, targets, target_top)
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/Main.py", line 1369:
    jobs.run(postfunc = jobs_postfunc)
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Taskmaster/Job.py", line 125:
    postfunc()
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/Main.py", line 1366:
    SCons.SConsign.write()
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/SConsign.py", line 131:
    syncmethod()
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/dblite.py", line 195:
    with self._open(self._tmp_name, "wb", opener=self.opener) as f:
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/dblite.py", line 177:
    return os.open(path, flags, mode=self._mode)
Exception ignored in: <function _Dblite.__del__ at 0x10529b240>
Traceback (most recent call last):
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/dblite.py", line 184, in __del__
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/dblite.py", line 181, in close
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/dblite.py", line 195, in sync
  File "/Users/frederik/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/dblite.py", line 177, in opener
FileNotFoundError: [Errno 2] No such file or directory: '/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/.pio/build/esp32-JoyIt/.sconsign311.tmp'
```

I don't know if this is intended or not, but after a failed build, the corresponding folder inside the .pio/build folder gets removed, so I cannot even do a post-mortem here.

I've seen other solutions on the web at least for this `.sconsign311.tmp` error, which included removing the directories I mentioned above, but as I said, none of that helped in the long run. I'd be glad if someone could help me fix this. Thank you!

fredlcore, 2024-03-01:
Here's my platformio.ini, just in case:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir = src/BSB_LAN

[env:Olimex-EVB]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32-evb
framework = arduino
build_src_filter = +<*>
upload_speed = 460800
board_build.partitions = min_spiffs.csv
build_flags = -DDS218 -DBSB_BUS -Werror=tautological-compare -Wtype-limits -Wimplicit-fallthrough
; -DCORE_DEBUG_LEVEL=5
upload_port = /dev/cu.*usb*serial*
build_type = debug
monitor_filters = esp32_exception_decoder, time
monitor_speed = 115200
lib_deps = arduino-libraries/Ethernet, miq19/eModbus

[env:due-PPS]
platform = atmelsam
board = due
framework = arduino
build_flags = 
	-DPPS_BUS
	-DDS213
lib_deps = 
	arduino-libraries/Ethernet@^2.0.1

[env:esp32-JoyIt]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder, time
build_src_filter = +<*>
upload_port = /dev/cu.usbserial*
upload_speed = 460800
board_build.partitions = min_spiffs.csv
build_flags = -DNODEMCU
lib_deps = 
	arduino-libraries/Ethernet
```

romainfd, 2024-03-03:
+1 I am seeing the same error `FileNotFoundError: [Errno 2] No such file or directory: '/[...path...]/.pio/build/nanoTag/.sconsign311.tmp'`

subhashisbhowmik, 2024-04-15:
+1 Same issue all of a sudden

DonnyVerduijn, 2024-04-20:
> What I figured out solves the problem most of the time is to completely remove the `.pio` folder (although "Full Clean" also removes all the contents within `.pio` it doesn't remove the folder itself) **and** before doing any build run, _save_ the `platformio.ini` file. Even with no changes, this starts the dependency run which would otherwise run when starting the build run. But the dependencies might not be completely downloaded/installed yet, so the build then fails. And with more failed builds, the matter seems to get worse. Sometimes, for example, I had `MyScript.ino.cpp`, `MyScript 1.ino.cpp` and `MyScript 2.ino.cpp` in my `src` folder (which should have been placed in the `build` folder above the `src` directory. Only when I figured out that they ended up in my `src` folder, the build run succeeded. So there's something not quite right in the way of the timing here, I guess, but the above mentioned procedure did the trick for me, at least for now...

Can confirm this. I wasn't able to pinpoint the root of the problem, but maybe someone can tell us when these errors started happening, so we can track it down in this repository.

fredlcore, 2024-04-20:
What exactly is it that you cannot confirm? I can reproduce here that after doing a full clean or `rm .pio`, the dependencies are not immediately pursued. I have to save `platformio.ini` in order to do so. Otherwise it will happen while the build run is being executed. This seems to be at least part of the problem on my side.

DonnyVerduijn, 2024-04-20:
I have to take a closer look tomorrow. If saving `platformio.ini` after the dependencies have downloaded, is reproducing a working scenario, that would help us a lot in the right direction.

fredlcore, 2024-04-20:
Maybe I wasn't expressing myself accurately, what I mean is this:
1. I encounter the problem
2. I remove the `.pio` folder
3. Then I have to save `platformio.ini` in order for
4. the dependencies begin to download.

If I omit step 3, the dependencies will only start to download upon starting the build process, so that compilation and downloading occurs in parallel.

closed 2025-05-06 by stale[bot]

HydroSausager, 2026-02-06:
issue is present

Teehee2, 2026-02-06:
> Maybe I wasn't expressing myself accurately, what I mean is this:
>
>     1. I encounter the problem
>
>     2. I remove the `.pio` folder
>
>     3. Then I have to save `platformio.ini` in order for
>
>     4. [...]

This method seems to have solved it.