platformio/platformio-vscode-ide #4028: UnknownPackageError: Could not find the package

HiImCris, 2024-04-14:
At startup and when trying to build a project, I get the following error message:

> UnknownPackageError: Could not find the package with 'earlephilhower/tool-pioasm-rp2040-earlephilhower @ ^5.100300.220714' requirements for your system 'windows_amd64'

I don't know how to solve it. Before that, I had to enable long path names. So that's where I'm now.

HiImCris, 2024-04-15:
It seems the code I downloaded refers to a repository that no longer exists. It builds without that reference, but I cannot test the results right now. So, for now, it's solved.

Erstadan, 2025-02-10:
@HiImCris How did you manage to build without that reference? I am unable to as far as I can see