platformio/platformio-vscode-ide #4081: Start PIO Home Server

elenonolasco, 2024-07-03:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.90.2
  PIO IDE: v3.3.3
  System: Windows_NT, 10.0.22000, x64

  %23 Exception
  ```
  Error: TimeoutError: Traceback (most recent call last):
  File "C:\Users\Eleno\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  %23 pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\Users\Eleno\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Eleno\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\Eleno\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Eleno\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Eleno\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Eleno\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Eleno\.platformio\penv\Lib\site-packages\platformio\home\cli.py", line 92, in cli
    run_server(
  File "C:\Users\Eleno\.platformio\penv\Lib\site-packages\platformio\home\run.py", line 89, in run_server
    uvicorn.run(
  File "C:\Users\Eleno\.platformio\penv\Lib\site-packages\uvicorn\main.py", line 575, in run
    server.run()
  File "C:\Users\Eleno\.platformio\penv\Lib\site-packages\uvicorn\server.py", line 65, in run
    return asyncio.run(self.serve(sockets=sockets))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Eleno\.platformio\python3\Lib\asyncio\runners.py", line 189, in run
    with Runner(debug=debug) as runner:
  File "C:\Users\Eleno\.platformio\python3\Lib\asyncio\runners.py", line 59, in __enter__
    self._lazy_init()
  File "C:\Users\Eleno\.platformio\python3\Lib\asyncio\runners.py", line 137, in _lazy_init
    self._loop = events.new_event_loop()
                 ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Eleno\.platformio\python3\Lib\asyncio\events.py", line 806, in new_event_loop
    return get_event_loop_policy().new_event_loop()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Eleno\.platformio\python3\Lib\asyncio\events.py", line 695, in new_event_loop
    return self._loop_factory()
           ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Eleno\.platformio\python3\Lib\asyncio\windows_events.py", line 315, in __init__
    super().__init__(proactor)
  File "C:\Users\Eleno\.platformio\python3\Lib\asyncio\proactor_events.py", line 639, in __init__
    self._make_self_pipe()
  File "C:\Users\Eleno\.platformio\python3\Lib\asyncio\proactor_events.py", line 784, in _make_self_pipe
    self._ssock, self._csock = socket.socketpair()
                               ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Eleno\.platformio\python3\Lib\socket.py", line 645, in socketpair
    ssock, _ = lsock.accept()
               ^^^^^^^^^^^^^^
  File "C:\Users\Eleno\.platformio\python3\Lib\socket.py", line 294, in accept
    fd, addr = self._accept()
               ^^^^^^^^^^^^^^
TimeoutError: timed out

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

C:\Users\Eleno\.platformio\penv\Lib\site-packages\platformio\__main__.py:132: RuntimeWarning: coroutine 'Server.serve' was never awaited
  exit_code = int(str(exc)) if str(exc).isdigit() else 1
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
Exception ignored in: <function BaseEventLoop.__del__ at 0x000001AEF75A9EE0>
Traceback (most recent call last):
  File "C:\Users\Eleno\.platformio\python3\Lib\asyncio\base_events.py", line 694, in __del__
    self.close()
  File "C:\Users\Eleno\.platformio\python3\Lib\asyncio\proactor_events.py", line 696, in close
    self._close_self_pipe()
  File "C:\Users\Eleno\.platformio\python3\Lib\asyncio\proactor_events.py", line 776, in _close_self_pipe
    self._ssock.close()
    ^^^^^^^^^^^
AttributeError: 'ProactorEventLoop' object has no attribute '_ssock'

	at c:\Users\Eleno\.vscode\extensions\platformio.platformio-ide-3.3.3-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:11001
	at ChildProcess.i (c:\Users\Eleno\.vscode\extensions\platformio.platformio-ide-3.3.3-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4621)
	at ChildProcess.emit (node:events:514:28)
	at ChildProcess.emit (node:domain:488:12)
	at maybeClose (node:internal/child_process:1105:16)
	at ChildProcess._handle.onexit (node:internal/child_process:305:5)
  ```

elenonolasco, 2024-07-03:
falha constante

tarcisiocipriano, 2024-10-02:
Conseguisse resolver?

closed 2025-04-26 by stale[bot]