platformio/platformio-vscode-ide #4105: PlatformIO-Test zeigt bei allen Modulen gleichen Fehler

ottomit4t70, 2024-08-19:
Type: <b>Bug</b>

Testlauf  wird mit Fehler abgebrochen. Folgende Meldung wird angezeigt:

Error: Usage: python -m platformio test [OPTIONS]
Try 'python -m platformio test -h' for help.

Error: Got unexpected extra arguments (CHANGELOG.md crowdin.yml data hal include lib LICENSE Neuer Ordner.code-workspace platformio.ini platformio_override-template.ini platformio_override.ini README.md src test tools user_setups)

Extension version: 3.3.3
VS Code version: Code 1.92.2 (fee1edb8d6d72a0ddff41e5f71a671c23ed924b9, 2024-08-14T17:29:30.058Z)
OS version: Windows_NT x64 10.0.22631
Modes:

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|AMD Ryzen 5 3400G with Radeon Vega Graphics     (8 x 3700)|
|GPU Status|2d_canvas: enabled<br>canvas_oop_rasterization: enabled_on<br>direct_rendering_display_compositor: disabled_off_ok<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>opengl: enabled_on<br>rasterization: enabled<br>raw_draw: disabled_off_ok<br>skia_graphite: disabled_off<br>video_decode: enabled<br>video_encode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled<br>webgpu: enabled<br>webnn: disabled_off|
|Load (avg)|undefined|
|Memory (System)|29.95GB (20.86GB free)|
|Process Argv|--crash-reporter-id ab11826f-5030-41cf-9cd8-7130a2cd924d|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368cf:30146710
vspor879:30202332
vspor708:30202333
vspor363:30204092
vscod805cf:30301675
binariesv615:30325510
vsaa593cf:30376535
py29gd2263:31024239
c4g48928:30535728
azure-dev_surveyone:30548225
962ge761:30959799
pythongtdpath:30769146
pythonnoceb:30805159
asynctok:30898717
pythonregdiag2:30936856
pythonmypyd1:30879173
2e7ec940:31000449
pythontbext0:30879054
dsvsc016:30899300
dsvsc017:30899301
dsvsc018:30899302
cppperfnew:31000557
dsvsc020:30976470
pythonait:31006305
dsvsc021:30996838
f3je6385:31013174
pythoncenvpt:31062603
a69g1124:31058053
dvdeprecation:31068756
dwnewjupytercf:31046870
impr_priority:31102340
nativerepl1:31104043
refactort:31108082
pythonrstrctxt:31112756
flighttreat:31119336
wkspc-onlycs-t:31111718
wkspc-ranged-c:31118571

```

</details>

adabru, 2025-02-12:
I'm getting the same error:

```
Error: Usage: python -m platformio test [OPTIONS]
Try 'python -m platformio test -h' for help.

Error: Got unexpected extra arguments (install_dependencies.py lib platformio.ini prebuild.py README.md README.txt src test)
```

PlatformIO IDE v3.3.4 
VS Code 1.97.0
Windows 11

adabru, 2025-02-12:
Alright, so I tried with the example project from https://github.com/platformio/platformio-examples/tree/develop/unit-testing/calculator, which was running fine. I compared my project with the example and the difference was that the example had additional test_*/ folders inside <project_root>/test/. So I created a new folder in test/ and placed my tests there. After that, it worked.

closed 2025-05-08 by stale[bot]