platformio/platformio-vscode-ide #4174: Test Results tab does not display carriage returns on MacOS

nooj, 2024-11-13:
When running tests on MacOS (14.6 and prior), the "Test Results" tab does not move the cursor to the left at the end of a line (ie carriage return '\r').  It only performs a linefeed ('\n').

See screenshot at this link: https://community.platformio.org/t/output-terminal-issues-during-testing/43944

The "Terminal" tab correctly displays identical compiler/linker output as the "Test Results" tab.  Presumably, whatever code the "Terminal" tab has to handle "\n" on Mac as "\r\n" does not happen in the "Test Results" code.

Here is a project which shows the error: https://github.com/platformio/platformio-examples/tree/develop/unit-testing/calculator

conr2286, 2025-03-02:
#4174 is still with us on Mar 2, 2025.  Screenshot attached.

PlatformIO 3.3.4 (2025-01-11)
Visual Studio Code Version 1.97.2
MacBook Air M2 (ARM)
MacOS:  Sequoia 15.3.1

As noted above, the problem does not occur in the TERMINAL tab when I click the test icon on the bottom of the screen, only in TEST RESULTS when I click the test icon in the sidebar (visible in screenshot).  Please contact me (conr2286 AT gmail) if more info needed.

[image]

gdela, 2025-05-10:
I have exactly the same problem, but on Ubuntu 24.10, Visual Studio Code 1.99.3, PlatformIO 3.3.4 and using Google Test framework:

[image]

jfuerth, 2025-08-09:
This is still an issue for me today on PlatformIO Extension 3.3.4 + VS Code 1.102.3 + Ubuntu 24.04.2.

[image]