platformio/platformio-vscode-ide #4202: Suddenly vs-code won't prompt me to reboot!

mag1cmao, 2024-12-29:
Computer: MacBook Pro 
Chip: Apple M3 Pro
macOS: Sequoia 15.2 

platformio.ini information：
[env:STC89C52RC]
platform = intel_mcs51
board = STC89C52RC
upload_protocol = stcgal
monitor_speed = 115200
upload_speed = 256000
upload_port = /dev/cu.wchusbserial210

I'm using vs-code to learn the STC89C52RC microcontroller with the PlatformIO plugin, which is very handy.
But today I'm having a problem, as shown in the picture, before it would prompt me to reboot, now it keeps showing Uploding ..., only after I power off. May I ask what is the situation?

I want to be prompted to reboot, not I don't know what to do and go try rebooting.
I have made no changes to the settings, configuration, and I want to be prompted to reboot. 
Thank you all so much for your help.

mag1cmao, 2024-12-29:
[image]
[image]

mag1cmao, 2024-12-30:
The problem remains unresolved, but in my attempts to solve it, I have discovered another phenomenon.
When I burn manually, I get a reboot prompt with the following command:
stcgal -P stc89 -p /dev/tty.wchusbserial210 /Users/shawnmao/Documents/PlatformIO/Projects/Test/.pio/build/STC89C52RC/firmware.hex

However, there is still no prompt using PlatformIO, and I'm wondering why there was a prompt a few days ago, has anyone else had a similar problem?

[image]
[image]
[image]
[image]

sizhuxuan, 2025-02-20:
我也遇到了这个问题，build就没有问题，当upload的时候，输出到Cycling power: done，开发板直接断掉了，此时切换电源键也不起作用。只有拔掉usb重插才显示正常状态。我尝试手动将platformio build后的 .hex烧录进开发板是没有问题的。
你的问题解决了吗？[图片]

mag1cmao, 2025-02-22:
我没有解决。。。。 

林隐月
***@***.***

   

  sizhuxuan  思竹轩 left a comment   发表评论(platformio/platformio-vscode-ide#4202)
（平台IO/平台IO-VSCODE-IDE#4202）
 
 
直接回复此电子邮件，在 GitHub 上查看，或取消订阅。

closed 2025-05-08 by stale[bot]