platformio/platformio-vscode-ide #4215: Error 404 in microsoft marketplace

arsenier, 2025-01-10:
I wanted to install PlatformIO on new vscode install, however I couldn't find extension in marketplace

[image]

After that I went to the Microsoft Extension marketplace ant tried to find it there, also no luck :/

[image]

Then I tried to use direct links from installation page. They lead to 404 page:

[image]

https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide

[image]

What happened and how can I install extension?

anthonyvallone, 2025-01-10:
Im facing the same issue....

sivar2311, 2025-01-10:
See [IDE Extension has disappeared from official VSCode Marketplace?](https://community.platformio.org/t/ide-extension-has-disappeared-from-official-vscode-marketplace/45202/8)

arsenier, 2025-01-10:
Was it an AI response? :)

arsenier, 2025-01-10:
https://github.com/platformio/platformio-vscode-ide/issues/4215#issuecomment-2582549994
That worked, pio installed

arsenier, 2025-01-10:
However there is another issue, when I try to create new project it gives an error and following github issue text. There are a lot of issues with this error as I saw, so I don't want to create a new one. What is it?

Start PIO Home Server

%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.96.2
  PIO IDE: v3.3.3
  System: Linux, 6.9.3-76060903-generic, x64

  %23 Exception
  ```
  Error: TypeError: Traceback (most recent call last):
  File "/home/arsenier/.platformio/penv/lib/python3.12/site-packages/platformio/__main__.py", line 103, in main
    cli()  %23 pylint: disable=no-value-for-parameter
    ^^^^^
  File "/home/arsenier/.platformio/penv/lib/python3.12/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arsenier/.platformio/penv/lib/python3.12/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/arsenier/.platformio/penv/lib/python3.12/site-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/arsenier/.platformio/penv/lib/python3.12/site-packages/click/core.py", line 1697, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arsenier/.platformio/penv/lib/python3.12/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arsenier/.platformio/penv/lib/python3.12/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arsenier/.platformio/penv/lib/python3.12/site-packages/platformio/home/cli.py", line 92, in cli
    run_server(
  File "/home/arsenier/.platformio/penv/lib/python3.12/site-packages/platformio/home/run.py", line 66, in run_server
    if not os.path.isdir(contrib_dir):
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen genericpath>", line 42, in isdir
TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /home/arsenier/.vscode/extensions/platformio.platformio-ide-3.3.3/node_modules/platformio-node-helpers/dist/index.js:1:10939
	at ChildProcess.i (/home/arsenier/.vscode/extensions/platformio.platformio-ide-3.3.3/node_modules/platformio-node-helpers/dist/index.js:1:4559)
	at ChildProcess.emit (node:events:518:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1104:16)
	at Socket.<anonymous> (node:internal/child_process:456:11)
	at Socket.emit (node:events:518:28)
	at Socket.emit (node:domain:489:12)
	at Pipe.<anonymous> (node:net:343:12)
  ```

The-EG, 2025-01-10:
This extension is still not listed in the VS Code marketplace, which is what this issue was originally opened for. I'd suggest leaving this open, so that others have a better chance of seeing this info and hopefully not creating duplicate issues.

rx-ted, 2025-01-10:
I have the same question. When will the PlatformIO extension be released or updated in the VS Code Marketplace?

sivar2311, 2025-01-10:
@rx-ted See https://community.platformio.org/t/ide-extension-has-disappeared-from-official-vscode-marketplace/45202/8

anthonyvallone, 2025-01-10:
@ivankravets its appeared back in the vs code extensions but it says its not available for apple silicon?

dosipod, 2025-01-10:
Also the same issue in codespaces 
[image]

ivankravets, 2025-01-11:
Duplicate of https://github.com/microsoft/vsmarketplace/issues/1114

The new version of extension has been published.

closed 2025-01-11 by ivankravets