platformio/platformio-vscode-ide #4284: Possibility to use Platformio IDE without depending on the ms-vscode.cpptools extension

kr4fty, 2025-04-08:
I currently use VSCode, VSCodium, under Linux, and when I install PlatformIO IDE, `cpptools` is installed as a dependency. So far, so good.
When I restart and try to open my project, cpptools gives me the following error message:

> The C/C++ extension may be used only with Microsoft Visual Studio, Visual Studio for Mac, Visual Studio Code, Azure DevOps, Team Foundation Server, and successor Microsoft products and services to develop and test your applications.

Is there a way to use PlatformIO IDE without necessarily having to use `cpptools`? For example, using `clangd`, `clang-forma`t and `clang-tidy`

psolyca, 2025-04-08:
https://github.com/platformio/platformio-vscode-ide/issues/1940

I disable C/C++ extention so platformio is deactivated also so use the command line to build, upload and monitor inside VSCodium.
You should be able to activate the platformio environnement in VSCodium

redocrehtona, 2025-07-15:
Bump, just making sure this doesn't get closed

z-jxy, 2025-08-14:
+1 for this

ivankravets, 2025-10-06:
See https://github.com/platformio/platformio-vscode-ide/issues/1802#issuecomment-3370225987

ahan, 2025-12-03:
another alternative solution is downgrade ms-vscode.cpptools to version 1.24.4

https://github.com/microsoft/vscode-cpptools/releases/tag/v1.24.4