platformio/platformio-vscode-ide #4325: PlatformIO IDE Install Problems

SalmonPenny323, 2025-06-17:
I am on Windows 11, coding on Visual Studio Code. 

When I install PIO, I go to the sidebar icon where it says "Initializing PlatformIO Core..." In the bottom right, a notification says, "PlatformIO Installer: Installing PlatformIO Core." Another popup says, "PlatformIO: Can not find working Python 3.6+ Interpreter. Please install the latest Python 3 and restart VSCode." I click the "I have Python" button, and I insert the Python interpreter file path. After it finishes installing, VSCode needs to restart. After restarting VSCode, the process repeats. 

These screenshots show the order of how stuff happens. If there is anything else needed to fix this, let me know.

[image]
[image]
[image]

SalmonPenny323, 2025-06-28:
I downloaded it from the official website. After searching it up, some people said to download an older version of Python, which I did. But neither the new version nor the older version worked.

ivankravets, 2025-06-30:
Do you use any espionage software? (antivirus tools, etc.)