platformio/platformio-vscode-ide #4328: PlatformIO cant be install as a extension on MacOS on Cursor Editor

Peeeeteer, 2025-06-22:
Not sure if I should be posting this here, 
but you cannot install PlatformIO as a Extension inside the Cursor Editor on a MacOS devices, but can be install on a Windows Device.

Mac + VSCode + PlatformIO works fine?
So not sure why it does not work inside Cursor as they are basically the same thing?

I did send an email to the Cursor Team, but add this here aswell just incase.

Peeeeteer, 2025-07-25:
> That's not a good method. It will overwrite existing settings in Cursor (aside from the fact it requires installing something else to install what we want).

Well yes,
but its the only one that works now, there is literally no other way to have it be in cursor

davidgomes, 2025-08-17:
Hey everyone, can you please try this in Cursor? [https://open-vsx.org/extension/DavidGomes/platformio-ide](https://open-vsx.org/extension/DavidGomes/platformio-ide)

I have been able to get this to work on my local Cursor for any PlatformIO project.

What I did was:
* Make this require the Clang CPP extension (so you should get a notification to install it if you haven't already, and this extension won't even load until you do that)
* For every PlatformIO project, the extension now generates a `compile_commands.json` file in the project directory so that the Clang CPP extension works properly